Xilinx/PYNQ #223: Vivado TCL script fails in 2016.2

baronep, 2016-10-21:
I guess the script was created in 2016.1. When I run "make" in Vivado I get the following error ...

[vivado.log.txt](https://github.com/Xilinx/PYNQ/files/543232/vivado.log.txt)

yunqu, 2016-10-21:
Yes the script was created for Vivado 2016.1. For 2016.2, you can modify line 23 of base.tcl to:
```tcl
set scripts_vivado_version 2016.2
```

However, we have not tested this completely. But you should be able to build the bitstream.

brad0taylor, 2016-10-26:
I'm having the same issue with 2016.3 and when I change the version I get the following error, which prevents me from rebuilding the bitstream.

> ERROR: [BD 5-216] VLNV <xilinx.com:ip:microblaze:9.6> is not supported for the current part. The latest supported version for this part is:10.0

brad0taylor, 2016-10-26:
btw I'm not very familiar with the process here. Should I have opened a new issue about this?

yunqu, 2016-10-31:
For 2016.2, you can also upgrade an existing project.

1.Unzip attached file and open project in Vivado (2016.2)
2.Upgrade IP when asked, and regenerate the bitstream

closed 2016-10-31 by yunqu