Xilinx/PYNQ #224: TCL script fails in Vivado 2016.3

brad0taylor, 2016-10-26:
The release tcl script fails with Vivado 2016.3. When  I change the version in the script,  I get the following error, which prevents me from rebuilding the bitstream.

> ERROR: [BD 5-216] VLNV xilinx.com:ip:microblaze:9.6 is not supported for the current part. The latest supported version for this part is:10.0

brad0taylor, 2016-10-28:
I was able to work around the script issue withe the following instructions from Benedikt in the Pynq user forum.  (I did get the error message below). Not sure how to generate the tcl script or I would post it here. 

You can use the these steps to upgrade it to 2016.x:
1.Unzip attached file and open project in Vivado (2016.x)
2.Upgrade IP when asked
3.If there is an error message that says "unable to read 720p_edid.txt", you can copy the file manually using Vivado Tcl Console:
(Windows) copy ./base.ipdefs/ip_0/dvi2rgb_v1_6/src/720p_edid.txt ./base.src/sources_1/bd/system/ipshared/050c/src
(Linux) cp ./base.ipdefs/ip_0/dvi2rgb_v1_6/src/720p_edid.txt ./base.src/sources_1/bd/system/ipshared/050c/src

yunqu, 2016-10-28:
You can use the following TCL file for the overlay class. 

`<project_folder>\<project_name>.srcs\sources_1\bd\system\hw_handoff\system_bd.tcl`

However, if you want to use the TCL file to generate the bitstream/hdf, you need to add the following to the end of that tcl file (suppose your project is called `base`):

```tcl
# Additional steps to get to bitstream
# generate toplevel wrapper files
make_wrapper -files [get_files ./base/base.srcs/sources_1/bd/system/system.bd] -top

add_files -norecurse ./base/base.srcs/sources_1/bd/system/hdl/system_wrapper.v
update_compile_order -fileset sources_1
update_compile_order -fileset sim_1
add_files -fileset constrs_1 -norecurse ./src/constraints/top.xdc

# replace top wrapper with custom top.v
add_files -norecurse ./src/top.v
update_compile_order -fileset sources_1
set_property top top [current_fileset]
update_compile_order -fileset sources_1

# call implement
launch_runs impl_1 -to_step write_bitstream -jobs 4
wait_on_run impl_1

# This hwardware definition file will be used for microblaze projects
file mkdir ./base/base.sdk
write_hwdef -force  -file ./base/base.sdk/base.hdf
file copy -force ./base/base.sdk/base.hdf ../../sdk/

# move and rename bitstream to final location
file copy -force ./base/base.runs/impl_1/top.bit ../../bitstream/base.bit

```

An example can be found at: https://github.com/Xilinx/PYNQ/blob/master/python/pynq/bitstream/base.tcl

closed 2016-10-31 by yunqu

promach, 2018-01-11:
@brad0taylor you mean [https://groups.google.com/forum/#!topic/pynq_project/qGS7m4_HGyY](url) ?

promach, 2018-01-11:
@yunqu [https://github.com/Xilinx/PYNQ/blob/master/python/pynq/bitstream/base.tcl](url) returns 404 NOT FOUND status

yunqu, 2018-01-11:
@promach  Thanks. 
The reason is that we have changed the folder structure starting from v2.0 release. The path you showed up there was the old path. The current path is https://github.com/Xilinx/PYNQ/blob/master/boards/Pynq-Z1/base/base.tcl

promach, 2018-01-12:
Why am I having the following screenshot CRITICAL WARNING when I open the implementation design ?

[image]

promach, 2018-01-12:
@brad0taylor I have modified the _base.tcl_ file accordingly for Vivado 2017.2 version. Feel free to have a look
[https://gist.github.com/promach/b26c8ecaf983bdd4f02042b5957c7a60](url)