Xilinx/PYNQ #225: HDMI input fails with google chromecast input

brad0taylor, 2016-10-27:
When I connect the PYNQ HDMI input to a google chromecast device, it appears to be capturing a blank/black screen. I'm using the 5_base_overlay_video.ipynb page for testing. It captures and displays video from my notebook fine.  Not sure if this is a configuration issue or perhaps its just not compatible, but I did want to record the issue.

brad0taylor, 2016-10-31:
Also, the HDMI port does seem to work fine with my old Apple TV.

plysaght, 2016-10-31:
You have got to love it when the new tech works with the older stuff, Brad

BTW Liam Madden just got his kit today so you are in good company

-P

This email and any attachments are intended for the sole use of the named recipient(s) and contain(s) confidential information that may be proprietary, privileged or copyrighted under applicable law. If you are not the intended recipient, do not read, copy, or forward this email message or any attachments. Delete this email message and any attachments immediately.

schelleg, 2017-01-16:
We have not seen fix for chromecast - I'll close for now, and put the help wanted label on here in case community wants to tackle this use case.

closed 2017-01-16 by schelleg