Xilinx/PYNQ #226: Direct Ethernet connection not working

brad0taylor, 2016-10-27:
I'm having some difficulty connecting to the notebook server with my PYNQ board.  Here are the symptoms:

* It usually seems to book ok to the notebook screen when etherent is connected to my local server. 
* Cannot connect to http://192.168.2.99:9090 with a direct ethernet cable after power up or reset
   *  Ping from PC to 192.168.2.99 fails
   *  Fails with multiple cables
   *  My windows PC does recognize that it is connected to an ethernet device at 1 gbs.. 
 *  after connecting a serial port to PYNQ
   *  pinging my PC from PYNQ fails with 'network unreachable'
   *  pinging 192.168.2.99 from PYNQ works

Here is the boot log from reset
[pynq_boot.txt](https://github.com/Xilinx/PYNQ/files/554803/pynq_boot.txt)

PeterOgden, 2016-10-27:
When directly connecting you need to ensure that the windows computer is configured to be on the same subnet as pynq. The instructions on how to do this are at http://pynq.readthedocs.io/en/latest/17_appendix.html#assign-your-laptop-pc-a-static-ip-address. If it still doesn't work, can you please provide the output of ipconfig on the windows machine and ifconfig on the pynq?

brad0taylor, 2016-10-27:
Thank you Peter - it works fine now. You might note that the PYNQ pictorial quickstart that ships doesn't mention this requirement Thanks for helping me out.

closed 2016-10-27 by brad0taylor