Xilinx/PYNQ #283: How to test grove oled example ?

Pillar1989, 2017-03-14:
I got a PYNQ board. 
[image]

I want test the grove oled example. But I meet a issue ? Please help me. That's my code.
```Python
# Load in the base Overlay
from pynq import Overlay
Overlay("base.bit").download()

from pynq.iop import PMOD_GROVE_G3
from pynq.iop import PMOD_GROVE_G4
from pynq.iop import Pmod_IIC

basic_font = \
[[0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00],
[0x00,0x00,0x5F,0x00,0x00,0x00,0x00,0x00],
[0x00,0x00,0x07,0x00,0x07,0x00,0x00,0x00],
[0x00,0x14,0x7F,0x14,0x7F,0x14,0x00,0x00],
[0x00,0x24,0x2A,0x7F,0x2A,0x12,0x00,0x00],
[0x00,0x23,0x13,0x08,0x64,0x62,0x00,0x00],
[0x00,0x36,0x49,0x55,0x22,0x50,0x00,0x00],
[0x00,0x00,0x05,0x03,0x00,0x00,0x00,0x00],
[0x00,0x1C,0x22,0x41,0x00,0x00,0x00,0x00],
[0x00,0x41,0x22,0x1C,0x00,0x00,0x00,0x00],
[0x00,0x08,0x2A,0x1C,0x2A,0x08,0x00,0x00],
[0x00,0x08,0x08,0x3E,0x08,0x08,0x00,0x00],
[0x00,0xA0,0x60,0x00,0x00,0x00,0x00,0x00],
[0x00,0x08,0x08,0x08,0x08,0x08,0x00,0x00],
[0x00,0x60,0x60,0x00,0x00,0x00,0x00,0x00],
[0x00,0x20,0x10,0x08,0x04,0x02,0x00,0x00],
[0x00,0x3E,0x51,0x49,0x45,0x3E,0x00,0x00],
[0x00,0x00,0x42,0x7F,0x40,0x00,0x00,0x00],
[0x00,0x62,0x51,0x49,0x49,0x46,0x00,0x00],
[0x00,0x22,0x41,0x49,0x49,0x36,0x00,0x00],
[0x00,0x18,0x14,0x12,0x7F,0x10,0x00,0x00],
[0x00,0x27,0x45,0x45,0x45,0x39,0x00,0x00],
[0x00,0x3C,0x4A,0x49,0x49,0x30,0x00,0x00],
[0x00,0x01,0x71,0x09,0x05,0x03,0x00,0x00],
[0x00,0x36,0x49,0x49,0x49,0x36,0x00,0x00],
[0x00,0x06,0x49,0x49,0x29,0x1E,0x00,0x00],
[0x00,0x00,0x36,0x36,0x00,0x00,0x00,0x00],
[0x00,0x00,0xAC,0x6C,0x00,0x00,0x00,0x00],
[0x00,0x08,0x14,0x22,0x41,0x00,0x00,0x00],
[0x00,0x14,0x14,0x14,0x14,0x14,0x00,0x00],
[0x00,0x41,0x22,0x14,0x08,0x00,0x00,0x00],
[0x00,0x02,0x01,0x51,0x09,0x06,0x00,0x00],
[0x00,0x32,0x49,0x79,0x41,0x3E,0x00,0x00],
[0x00,0x7E,0x09,0x09,0x09,0x7E,0x00,0x00],
[0x00,0x7F,0x49,0x49,0x49,0x36,0x00,0x00],
[0x00,0x3E,0x41,0x41,0x41,0x22,0x00,0x00],
[0x00,0x7F,0x41,0x41,0x22,0x1C,0x00,0x00],
[0x00,0x7F,0x49,0x49,0x49,0x41,0x00,0x00],
[0x00,0x7F,0x09,0x09,0x09,0x01,0x00,0x00],
[0x00,0x3E,0x41,0x41,0x51,0x72,0x00,0x00],
[0x00,0x7F,0x08,0x08,0x08,0x7F,0x00,0x00],
[0x00,0x41,0x7F,0x41,0x00,0x00,0x00,0x00],
[0x00,0x20,0x40,0x41,0x3F,0x01,0x00,0x00],
[0x00,0x7F,0x08,0x14,0x22,0x41,0x00,0x00],
[0x00,0x7F,0x40,0x40,0x40,0x40,0x00,0x00],
[0x00,0x7F,0x02,0x0C,0x02,0x7F,0x00,0x00],
[0x00,0x7F,0x04,0x08,0x10,0x7F,0x00,0x00],
[0x00,0x3E,0x41,0x41,0x41,0x3E,0x00,0x00],
[0x00,0x7F,0x09,0x09,0x09,0x06,0x00,0x00],
[0x00,0x3E,0x41,0x51,0x21,0x5E,0x00,0x00],
[0x00,0x7F,0x09,0x19,0x29,0x46,0x00,0x00],
[0x00,0x26,0x49,0x49,0x49,0x32,0x00,0x00],
[0x00,0x01,0x01,0x7F,0x01,0x01,0x00,0x00],
[0x00,0x3F,0x40,0x40,0x40,0x3F,0x00,0x00],
[0x00,0x1F,0x20,0x40,0x20,0x1F,0x00,0x00],
[0x00,0x3F,0x40,0x38,0x40,0x3F,0x00,0x00],
[... 240 lines cut ...]

oled = Python_Grove_OLED(PMODB,PMOD_GROVE_G3)
oled.clear()
oled.write('Hi from Python.')
del oled

#...... 
```
Output
```shell
xilinx@pynq:~/arduino$ sudo python3 oled_test.py 
[sudo] password for xilinx: 
Traceback (most recent call last):
  File "oled_test.py", line 311, in <module>
    oled = Python_Grove_OLED(PMODB,PMOD_GROVE_G3)
  File "oled_test.py", line 148, in __init__
    self._send_cmd(0xFD) 
  File "oled_test.py", line 171, in _send_cmd
    self.send([0x80,word])
  File "/home/xilinx/arduino/pynq/iop/pmod_iic.py", line 199, in send
    raise RuntimeError("Timeout when writing IIC.")
RuntimeError: Timeout when writing IIC.
xilinx@pynq:~/arduino$ 

```

yunqu, 2017-03-14:
Hi @Pillar1989 , I think you should connect your pmod grove OLED to G3 of the PYNQ grove adapter, and then connect the adapter to the PMODB connector (in order to use the Pmod IIC module). 

From the picture you have, it seems that you are using the Arduino IIC connector. If you want to use the Arduino shield, as shown in the picture, you should do (with proper importing, of course):
```python
oled = Grove_OLED(ARDUINO, ARDUINO_GROVE_IIC)
```

Pillar1989, 2017-03-14:
Thank you,  It's works. That's the full code.

```Python
from pynq import PL
from pynq.iop import Grove_OLED
from pynq.iop import PMODB
from pynq.iop import PMOD_GROVE_G3
from pynq.iop import ARDUINO
from pynq.iop import ARDUINO_GROVE_I2C
# Flush IOP
PL.reset()

oled = Grove_OLED(ARDUINO,ARDUINO_GROVE_I2C)
oled.clear()
oled.write('Hello from      Microblaze.')
del oled
```

closed 2017-03-14 by Pillar1989