Xilinx/PYNQ #284: Not working interrupt - Arduino with asyncio

simoamico94, 2017-03-16:
Hi, I was trying to use the interrupt mechanism but I had some troubles.  To exclude any sensor issues, I created a program for the Microblaze that raises an interrupt every second, expecting to catch it from Python. For the Python code, I basically copied from arduino_lcd18.py and I've also added the interrupt functions from the documentation (http://pynq.readthedocs.io/en/latest/13a_interrupts.html#async-function). However, the interrupt doesn't work: in jupyter, I expect to see an output every second (the interrupt frequency), but I get none. Can someone help me?

[interrupt_asyncio.zip](https://github.com/Xilinx/PYNQ/files/848665/interrupt_asyncio.zip)

I attached .c and .py files. Thanks in advance.

yunqu, 2017-04-13:
@simoamico94  Sorry for the delay. 

I checked the python code and C code. Nothing strange at the first glance. Can you also show the notebook code?

Also, what is the output? A running cell that never stops? 

Can you next time point to a location on Github so it would be easier for us to check your code?

simoamico94, 2017-04-28:
@yunqu in the notebook code I simply wrote: "ai = Arduino_Interrupt(3)" after doing the import, in order to run the python code. I put the Arduino_Interrupt file in the iop folder. 
Yes, the output is a running cell that never stops. I don't have a location on Github for this code, sorry!

schelleg, 2018-09-13:
closing old issue - please check out the pynq forum as well: http://www.pynq.io/support.html.

closed 2018-09-13 by schelleg