Xilinx/PYNQ #305: HDMI input stability with 1080P

briankincaid, 2017-06-20:
I am having trouble getting a consistent input lock to 1080P sources with the PYNQ board.  I have been able to connect to 720P reliably.  

Do you have any suggestions to 'tune' the DVI2RGB code for stability?  I am able to work with verilog and vhdl, but I would like some suggestion on where to start.

elodg, 2017-10-26:
@briankincaid,
I don't know if you still need help, but keep in mind that the Zynq-7020 PL resources (specifically the BUFIO clock buffers) are only spec'd to work at 600MHz, which limits the maximum pixel clock to 120MHz, lower than 1080P@60Hz. You can try 1080P@30Hz that is guaranteed to work.
That being said however, you can still try your luck. I am working on adding wizard-instantiable ILA primitives to the IP. In the mean time you can try looking at the 'pEyeSize' net with Vivado's logic analyzer feature. It should tell you how wide the valid data eye is on each channel.

schelleg, 2018-05-13:
closing out  - we updated documentation to recommend 720p for the PYNQ-Z1

closed 2018-05-13 by schelleg