Xilinx/PYNQ #388: Can not rebuild the Overlay

hillhao, 2017-07-29:
Hi ,

I run vivado 16.1 with the instruction to rebuild a new overlay. I open the project in the /Pynq-Z1/vivado/base file, and then run the base.tcl. But I got the following warning and errors. Actually, the opened project was incomplete. I can not do anything with this project to rebuild a new 
overlay application. Could you help me to check the problems? Thank you so much.

Best regards,
Yufeng Hao

CRITICAL WARNING: [BD 41-52] Could not find the abstraction definition specified by the vlnv: digilentinc.com:interface:tmds_rtl:1.0 
CRITICAL WARNING: [BD 41-181] Type specified by the VLNV: 'digilentinc.com:interface:tmds_rtl:1.0', cannot be found. Interface port: 'hdmi_in' cannot be created
CRITICAL WARNING: [BD 41-52] Could not find the abstraction definition specified by the vlnv: digilentinc.com:interface:tmds_rtl:1.0 
CRITICAL WARNING: [BD 41-181] Type specified by the VLNV: 'digilentinc.com:interface:tmds_rtl:1.0', cannot be found. Interface port: 'hdmi_out' cannot be created
ERROR: [BD 5-216] VLNV <xilinx.com:user:audio_direct:1.0> is not supported for the current part. 
ERROR: [Common 17-39] 'create_bd_cell' failed due to earlier errors.

    while executing
"create_bd_cell -type ip -vlnv xilinx.com:user:audio_direct:1.0 audio_direct_0 "
    (procedure "create_hier_cell_audio" line 47)
    invoked from within
"create_hier_cell_audio [current_bd_instance .] audio"
    (procedure "create_root_design" line 84)
    invoked from within
"create_root_design """
    (file "F:/xil_pro/Overlay/Pynq-Z1/bitstream/base.tcl" line 3678)

parimalp, 2017-07-29:
Please make sure that IP directory is located at the correct path as per the base.tcl file.

Parimal

totnine, 2017-08-10:
The ip_repo_paths in the base.tcl should be modified to what the same as your custom project

closed 2017-10-11 by yunqu