Xilinx/PYNQ #462: Consider updating to 16.04 LTS (Xenial)

japm48, 2017-09-23:
Ubuntu 15.10 was "EOL"ed in July 2016. I suggest updating to 16.04.x LTS (Xenial) as it will be supported at least until 2021.

I guess most changes should go to the list of packages in `multistrap.config` and AFAIK, kernel upgrade is not required.

https://wiki.ubuntu.com/Releases
https://www.ubuntu.com/info/release-end-of-life

PeterOgden, 2017-09-27:
We will definitely upgrade for the next image release and I'm hoping to get the necessary changes in the sdbuild flow earlier so that other people can generate appropriate images. Unfortunately I can't give a time frame at the moment.

japm48, 2017-09-29:
Ok, I guess this is enough promise to convince me to purchase the board :)
Close this if you feel it convenient.

closed 2017-11-06 by yunqu