Xilinx/PYNQ #463: i use browser to log in the board , but the password is invaild , it's not 'xilinx' ?

superFJH, 2017-09-28:


cathalmccabe, 2017-09-28:
Which browser are you using? Can you try using Chrome?

superFJH, 2017-09-30:
chrome ,and used on Linux , but when i try to use pynq on win10 ,it worked .....

superFJH, 2017-09-30:
oh , it can work on chrome , thanks. but why it can't be used on firefox ?

PeterOgden, 2017-09-30:
The is a problem with Firefox when the date on the board is more than 30 days in the past. We set the date on the image when we upload it but unless the board has a way through DHCP or NTP  to update the time it will drift. If you update the time on the board using `sudo date -s "YYYY/MM/DD hh:mm:ss"` this will allow you to use Firefox. There's more details on the problem here https://groups.google.com/forum/#!msg/pynq_project/l3bKT9VSAmc/-CruhNIOAQAJ;context-place=forum/pynq_project.

closed 2017-10-11 by yunqu