Xilinx/PYNQ #475: No HDMI input handshake

aasseman, 2017-11-03:
I'm having trouble getting HDMI input to work with `base.video.hdmi_in` in python (pynq API v2.0, had the same issue with 1.4):

- Pynq's HDMI input is not detected by devices, therefore devices do not start streaming video data. `base.video.hdmi_in` consequently times out and raises an exception.

- I managed to trick devices to start their HDMI output by using a DVI splitter (with HDMI-DVI adapters), connected to normal displays, and connecting one of the outputs of that splitter to the pynq.
I can then run all the video demos from the provided jupyter notebooks with 1080p 60fps input.

hasegaw, 2017-11-06:
I believe you would have faced one or both of these problems:

1)
PYNQ board doesn't have HDMI receiver LSIs. I experienced several PYNQ boards (and DVI2RGB IP in PYNQ) cannot sync with pixel clock provided by specific source devices (I doubt poor quality signals). In this case, PYNQ fails to start hdmi_in. Either of use of active HDMI cable or "retransmitting" the signal (using HDMI splitter, etc.) would resolve this issue.

I believe PYNQ-Z1 physical design needs to be changed for complete resolution. BTW I saw ZYBO Z7 boards have an LSI in HDMI RX.

2)
DVI2RGB accepts DVI data format only. Though the IP has DDC ROM to ask the source for DVI data format (on the HDMI port) so it should work for most cases, some sources ignore DDC and transmit in HDMI data format that DVI2RGB cannot handle. In this case, DVI2RGB can lock the pixel clock but no inputs from HDMI port (VDE always 0.)
Your "HDMI to DVI converter" would have resolved the situation by forcing DVI format.

This branch has the changes, to accept HDMI video format with audio, working fine with Nintendo Wii U console.
https://github.com/hasegaw/PYNQ/commits/edid_for_wiiu

I don't know if PYNQ will support HDMI data formats in future releases.

Takeshi

schelleg, 2018-05-13:
Yes, future boards (and frontend IP) will support HDMI properly.  On PYNQ-Z1, its really DVI

closed 2018-05-13 by schelleg