Xilinx/PYNQ #476: Place design Error (Implementation failed)

DevasenaInupakutika, 2017-11-06:
Synthesises successfully but throws "place design error" during implementation when we source the base.tcl file. 

Could you please let me know what might be missing?

Error screenshots attached below:

[image]
[image]

yunqu, 2017-11-07:
The first thing that I have noticed, is that you are using Vivado 2016.2. The `base.tcl` on image v2.0 is verified and tested based on Vivado 2016.1, while there is no guarantee that the same tcl file will work for 2016.2. Is there a reason why you must use Vivado 2016.2?

Also, we are working on a new release which supports Vivado 2017.2.

Hope this helps.

DevasenaInupakutika, 2017-11-08:
Thank you! There is no specific reason to use 2016.2. I had that installed on my system and have been working on either 2016.2 or 2016.4 of late, so tried with that version.  

That's great!

Also, ideally before reaching this implementation step, based on the condition in `base.tcl` file it should fail in the beginning itself (which it did). 

I changed the base.tcl script by making it run with 2016.2. It worked fine starting from sourcing the tcl files and building the PYNQ repository until Synthesis. 

The only reason I couldn't get is if it is version specific, it should fail initially even if I changed the tcl (one if condition block). 

Could you please comment or give input on this?

yunqu, 2017-11-08:
https://github.com/Xilinx/PYNQ/blob/master/boards/Pynq-Z1/base/base.tcl#L76

We do have that checking. But if you tried to change the version number, that is almost like a hack, so it may fail...

DevasenaInupakutika, 2017-11-11:
Ok, yeah! Thank you! It worked on 2016.1 version only.

closed 2017-11-11 by DevasenaInupakutika