Xilinx/PYNQ #499: sudo pip3.6 install --upgrade git+https://github.com/Xilinx/PYNQ.git won't work

onioncc, 2018-01-13:
Hi,

I am trying to run the following command:
sudo pip3.6 install --upgrade git+https://github.com/Xilinx/PYNQ.git

But I got the following error:

Collecting git+https://github.com/Xilinx/PYNQ.git                                                                                               
  Cloning https://github.com/Xilinx/PYNQ.git to /tmp/pip-5j6ud6cd-build                                                                         
    Complete output from command python setup.py egg_info:                                                                                      
    running egg_info                                                                                                                            
    creating pip-egg-info/pynq.egg-info                                                                                                         
    writing pip-egg-info/pynq.egg-info/PKG-INFO                                                                                                 
    writing top-level names to pip-egg-info/pynq.egg-info/top_level.txt                                                                         
    writing entry points to pip-egg-info/pynq.egg-info/entry_points.txt                                                                         
    writing dependency_links to pip-egg-info/pynq.egg-info/dependency_links.txt                                                                 
    writing manifest file 'pip-egg-info/pynq.egg-info/SOURCES.txt'                                                                              
    warning: manifest_maker: standard file '-c' not found                                                                                       
                                                                                                                                                
    reading manifest file 'pip-egg-info/pynq.egg-info/SOURCES.txt'                                                                              
    writing manifest file 'pip-egg-info/pynq.egg-info/SOURCES.txt'                                                                              
    Traceback (most recent call last):                                                                                                          
      File "<string>", line 1, in <module>                                                                                                      
      File "/tmp/pip-5j6ud6cd-build/setup.py", line 243, in <module>                                                                            
        subprocess.run(['systemctl' ,'restart', 'pl_server'])                                                                                   
    AttributeError: 'module' object has no attribute 'run'                                                                                      
    Restarting PL server                                                                                                                        
                                                                                                                                                
    ----------------------------------------     

Am I missing something?

Best Regards,
Cong

yunqu, 2018-01-14:
Thanks.

Is this based on image v2.0?

yunqu, 2018-02-02:
Thanks. I think the problem might be the SD card image is not up to date. Only upgrading PYNQ package from image v1.5 probably won't work.

closed 2018-02-26 by yunqu