Xilinx/PYNQ #584: xparameters lacking information for new uart.h/uart.c

jtbowman421, 2018-02-24:
With the new version of 2.1, A library was added for using uart-- uart.h and uart.c
However, xparameters has not been updated to show any instances of xuart, making this new library unusable. 

I have been working on a project where we were using version 2.0. In that version, we were using UartLite, which used to have a built-in baud rate of 9600. But with the updated version 2.1, it seems as though the built-in baud rate has been changed, and since updating our UART peripheral is no longer working. 

We would like to be able to use the new uart library, but it is nonfunctional, and it seems as though UartLite will also not work for us in the new update. A fix for the new uart files would be appreciated.

parimalp, 2018-02-24:
Uart is connected to Arduino connector and not to PMOD connectors. The baud rate set is 115200. The xparameters in Arduino bsp has the definition. Hope this helps to make progress.

Parimal

drichmond, 2018-02-24:
Is the rate 115200 or 9600?

http://pynq.readthedocs.io/en/v2.0/pynq_libraries/arduino.html

yunqu, 2018-04-14:
uart.c and uart.h should be fixed soon. Baud rate has been changed to 115200: https://github.com/Xilinx/PYNQ/blob/master/boards/Pynq-Z1/base/base.tcl#L2255

We will need to update the documentation for that.

yunqu, 2018-05-14:
This issue has been fixed in image v2.2.

closed 2018-05-14 by yunqu