Xilinx/PYNQ #585: Pynq v2.1 image

bgee1, 2018-02-26:
I am having trouble getting the board to boot up with the v2.1 image on windows 10. The Done light turns on but LEDs 0-3 never turn on.

yunqu, 2018-02-26:
Thanks for testing it out with Windows 10. Can you try to connect to USB terminal to see what error message? We need more information to identify the problem.

Btw, the first time it may take a little more time to boot up the board -- even longer for LEDs to flash.

bgee1, 2018-02-26:
Here a screen shot of the boot-up sequence. I hope that helps.
[image]

yunqu, 2018-02-26:
Hi, can you try downloading the image again, unzip it properly, and burn you SD image card? Have you also tried using a different SD card?

bgee1, 2018-02-26:
I downloaded the image again and this time everything worked. Not sure what happened the first time but thanks for your help!

closed 2018-02-26 by bgee1