Xilinx/PYNQ #603: modify kernel.config doesn't work

sonnyhcl, 2018-03-23:
I was trying to add some kernel config parameters by modifying `sdbuild/boot_configs/Pynq-Z1-defconfig/kernel.config` and it doesn't work. 
```makefile
# Zynq7000.makefile
${WORKDIR}/linux/.config: ${LINUX_CONFIG} | ${WORKDIR}/linux
	cp $< $@
```
I find that although `kernel.config` was copied to `$(WORKDIR)/linux` in `Zynq7000.makefile` , it's always overwrited by linux-xlnk build process and renamed as `.config.old`.

So what can I do to actually modify kernel config parameters and generate the bootable image?

PeterOgden, 2018-03-23:
That's the process we use when updating the kernel configuration. Can you post the config you are trying to use?

sonnyhcl, 2018-03-23:
I was trying to add kernel config parameters as below. After make process, I find that `sdbuild/.build/linux/.config`  is auto-generated and my config appears in `sdbuild/.build/linux/.config.old` . 

So what's the correct way to modify kernel config parameters?
```config
CONFIG_IPC_NS=y
CONFIG_UTS_NS=y
CONFIG_USER_NS=y
CONFIG_PID_NS=y
CONFIG_CGROUP_DEVICE=y
CONFIG_MEMCG=y
CONFIG_POSIX_MQUEUE=y
CONFIG_OVERLAY_FS=y
CONFIG_SECCOMP_FILTER=y
CONFIG_KEYS=y
CONFIG_SECCOMP=y
```

PeterOgden, 2018-03-23:
In this case .config has been regenerated because the options added have caused additional options to become available - when I replicated your config, the Linux build asked additional configuration questions. The  parameters you added are still in the new .config they've just been reordered according to the kernel's directory structure.

I tend to use `make ARCH=arm menuconfig` in the `.build/linux` directory to update the kernel configuration and then copy the new `.config` back into the boot_configs directory.

closed 2018-04-11 by yunqu