Xilinx/PYNQ #606: add bootargs mem=384M makes kernel failed

sl-zeng, 2018-03-30:
I try to use bootargs with 'mem=384M' to leave some space for my own kernel driver. However, it can not load the kernel correctly, and it seems something wrong with device-tree. I make an experiment that mem=500M is okay but mem=450 is failed. Is the PYNQ kernel inside the image demand such large mem space?
I used 'free -h' and see a free mem space of 280M. Is it possible for me to change the mem space manage by the PYNQ linux system?

PeterOgden, 2018-04-05:
If you want only a subset of the memory to be available to Linux then the best way is to set this through device tree. 
 1. Install device tree compiler on the board `sudo apt-get install device-tree-compiler`
 2. Decompile the existing device tree `dtc -I dtb -O dts /boot/devicetree.dtb > devicetree.dts`
 3. Edit the DTS file - you will find a `memory` section with `reg = <0x0 0x20000000>;` which is the configured memory range for the board. You can lower that by changing the upper limit as desired.
 4. Recompile the device tree `dtc -I dts -O dtb devicetree.dts > devicetree.dtb`
 5. Copy the new device tree back on to the boot partition - I recommend keeping a copy of the old one in case things don't work properly
 6. Unmount /boot `sudo umount /boot`
 7. Reboot the board

Hopefully this will do what you want.

sl-zeng, 2018-04-11:
Thanks for your reply.

I have successfully preserve the memory for linux by change this two args during boot stage:
fdt_high = 0x18000000
initrd_high = 0x18000000
In this way bootargs mem=384M can work.

And I have not tested your solution yet, but I think it shall work.
Best Regards

closed 2018-04-11 by sl-zeng