Xilinx/PYNQ #613: PYNQ Linux driver architecture

zhaozhixu, 2018-04-11:
Hi, I'm new to PYNQ and I'm trying to understand the architecture about PYNQ Linux drivers. But I can't understand why they can bypass Linux kernel and directly operate hardware registers.

From the source code of PYNQ, I find out that the drivers are all built upon some C code that directly calls assembly code to read/write hardware registers, without any interaction with Linux Loadable Kernel Module (LKM) system.

For example, the DMA driver is defined in PYNQ/pynq/lib/dma.py, which calls a shared object libdma.so, whose data structures and functions are defined in PYNQ/pynq/lib/_pynq/_apf/axidma_v9_1/xaxidma.h, which in the end includes some assembly code defined in PYNQ/pynq/lib/_pynq/_apf/include/xpseudo_asm_gcc.h. Nothing is mentioned about Linux kernel module.

How can that work? I mean it totally bypasses the whole Linux memory management system and so on? Is there any resources about this topic?

Thanks!

zhaozhixu, 2018-04-11:
UPDATE: I looked into more of dma.py, xlnk.py, overlay.py and mmio.py. It seems more clear to me.

In dma.py, there is 2 DMA class: LegacyDMA and DMA. 

LegacyDMA calls /usr/lib/libsds_lib.so to allocate host buffers and do the phy_addr-virt_addr convertion, and calls libdma.so to operate registers and do the transfers. 

The newer DMA needs an array allocated by Xlnk, which still calls /usr/lib/libsds_lib.so to allocate host buffers and do the phy_addr-virt_addr convertion. Mean while, it utilizes Python's built in mmap API to operate registers and do the transfers.

Is the above understanding right? And I'm looking for the source code compiling  /usr/lib/libsds_lib.so, but with no luck. Does anybody know where to find it? 

Thanks!

yunqu, 2018-04-11:
Hi,

`libsds_lib.so` is compiled on `libsds_lib.a` from the SDSoC tool suite, where source codes are probably not accessible to the public.

The APIs can be found at:
https://www.xilinx.com/support/documentation/sw_manuals/xilinx2015_2/sdsoc_doc/topics/api/reference_sdsoc_api.html

Thanks,

Rock

zhaozhixu, 2018-04-12:
@yunqu thanks, I'll look into it.

closed 2018-05-13 by schelleg