Xilinx/PYNQ #620: IPs not detected by Overlay when ZYNQ7 is in a hierarchical cell

quetric, 2018-04-23:
When placing the ZYNQ7 IP in a hierarchical cell, the TCL BD description is not properly parsed when loading the overlay, and IPs are not detected. See attached notebook:

[testhierarchy.tar.gz](https://github.com/Xilinx/PYNQ/files/1937307/testhierarchy.tar.gz)

Output when running the notebook cell is:
```
Test loading of overlay when PS7 is in hierarchy called CPU
Example Overlay loaded
{}
Test loading of overlay when PS7 is not in any hierarchy
Example Overlay loaded
{'axi_dma_0': {'phys_addr': 1077936128, 'addr_range': 65536, 'type': 'xilinx.com:ip:axi_dma:7.1', 'state': None, 'interrupts': {}, 'gpio': {}, 'fullpath': 'axi_dma_0', 'driver': <class 'pynq.lib.dma.DMA'>}, 'axi_dma_1': {'phys_addr': 1078001664, 'addr_range': 65536, 'type': 'xilinx.com:ip:axi_dma:7.1', 'state': None, 'interrupts': {}, 'gpio': {}, 'fullpath': 'axi_dma_1', 'driver': <class 'pynq.lib.dma.DMA'>}, 'example_0': {'phys_addr': 1136656384, 'addr_range': 65536, 'type': 'xilinx.com:hls:example:1.0', 'state': None, 'interrupts': {}, 'gpio': {}, 'fullpath': 'example_0', 'driver': <class 'pynq.overlay.DefaultIP'>}}
Test loading of overlay when PS7 is not in any hierarchy but DMAs are in a hierarchy called DMA
Example Overlay loaded
{'DMA/axi_dma_0': {'phys_addr': 1077936128, 'addr_range': 65536, 'type': 'xilinx.com:ip:axi_dma:7.1', 'state': None, 'interrupts': {}, 'gpio': {}, 'fullpath': 'DMA/axi_dma_0', 'driver': <class 'pynq.lib.dma.DMA'>}, 'DMA/axi_dma_1': {'phys_addr': 1078001664, 'addr_range': 65536, 'type': 'xilinx.com:ip:axi_dma:7.1', 'state': None, 'interrupts': {}, 'gpio': {}, 'fullpath': 'DMA/axi_dma_1', 'driver': <class 'pynq.lib.dma.DMA'>}, 'example_0': {'phys_addr': 1136656384, 'addr_range': 65536, 'type': 'xilinx.com:hls:example:1.0', 'state': None, 'interrupts': {}, 'gpio': {}, 'fullpath': 'example_0', 'driver': <class 'pynq.overlay.DefaultIP'>}}
```

This was observed on v2.1
```
xilinx@pynq:~/jupyter_notebooks$ cat ~/REVISION 
Release 2018_02_16 e5c8aeb
```
Vivado version is 2017.3
```
Vivado v2017.3 (64-bit)
SW Build: 2018833 on Wed Oct 4 19:58:07 MDT 2017
IP Build: 2016188 on Wed Oct 4 21:52:56 MDT 2017
```

yunqu, 2018-05-04:
Hi,

I think our assumption was the PS7 system is in the root level of the block design. I checked your tcl file; it looks like the PS7 system is wrapped in another hierarchy called CPU.

An easy work around is to just flatten that hierarchy so that the PS7 block is exposed at the top level. This will probably be the easiest way to make it work.

Thanks,

Rock

quetric, 2018-05-05:
Hi Rock,

I am aware that placing the PS7 in the toplevel is a valid workaround. It's just that AFAIK this is an undocumented requirement for overlay design. It should be documented somehow, or the parser should message the user and say what's wrong instead of failing silently as it does now.

PR #1208 (2021-07-19): Support PS block within a hierarchy
There is no change for designs with the PS block in the top level. For designs with the PS block within a hierarchy, the PS is added correctly to the `ip_dict`

Fixes #620 #1075