Xilinx/PYNQ #635: TUN TAP drivers

gourav4871, 2018-05-23:
Hello,

I want to use VPN on my PYNQ board. As there are no TUN module present in the default kernel provided [http://pynq.readthedocs.io/en/v2.1/getting_started.html#microsd-card-setup](url)
So I tried to re-compiled the kernel with a line, `CONFIG_TUN=m` added to the file `/PYNQ/sdbuild/boot_configs/Pynq-Z1-defconfig`. However, the compiled kernel also doesn't has any TUN module. 
`xilinx@pynq:~$ modinfo tun
modinfo: ERROR: Module tun not found.
`
Can someone tell what am I doing wrong ?
Kind regards,
Gourav

PeterOgden, 2018-05-24:
By using `CONFIG_TUN=m` you've compiled a kernel module which needs to be placed in the filesystem. In the build directory of your kernel you should find `tun.ko` which you can copy on to the board and load into the kernel using `insmod`.

Alternatively you may be able use `CONFIG_TUN=y` and update the uImage file in the boot partition which will compile TUN directly into the kernel.

gourav4871, 2018-05-24:
Now I am able to load the `tun.ko`
Thank you for the help.