Xilinx/PYNQ #641: VectorBlox MXP PYNQ question

promach, 2018-06-14:
@yunqu @schelleg @PeterOgden @vanjoe

I am now working on [VectorBlox MXP implementation on PYNQ](https://github.com/VectorBlox/PYNQ/blob/vectorblox/python/pynq/iop/mxp_filter.py#L43)

Why would sobel (edge detection) filter computation require [mmio](https://github.com/VectorBlox/PYNQ/blob/vectorblox/python/pynq/mmio.py) (external hardware peripheral) ?

promach, 2018-06-14:
Someone told me the following. Could anyone confirm if this is correct ?

> this mxp_filter.py thing compiles the mxp filter (/mxp-src/...) and then reconfigures the "IOP" from a microblaze to this mxp thingy

> [request_iop documentation](https://pynq.readthedocs.io/en/v1.3/pynq.iop.html#pynq.iop.iop.request_iop)

> "PMODA" is just an ID for a slot of reconfigurable fabric which incidentally has PMOD A attached to it, which is why they named it so

> so request_iop takes this slot and programs it with whatever "mb_program" you passed, in the current case the freshly compiled MXP

> these places can be programmed with any bitstream, not just a microblaze. they replaced the microblaze with the freshly synthesized mxp. the docs didn't anticipate this use, they only consider that you might want to change a microblaze, but still leave it as a microblaze

Could I say that "filter.bin" is actually the freshly synthesized mxp ?

vanjoe, 2018-06-14:
Hi,

It's been  awhile since I touched this stuff, but i think filter.bin is a freshly compiled software program that is run on the microblaze. ( the microblaze controls the MXP if I recall correctly).  

Joel

promach, 2018-06-18:
@vanjoe 

For [mxp_filters.ipynb](https://gist.github.com/promach/24094b931979e4135c3c55813a000cf9) , why do I have this error `"ValueError: could not broadcast input array from shape (1555200) into shape (0)"` ?

vanjoe, 2018-06-21:
Hi @promach 

I'm not going to be too helpful for the question regarding the Image() function, I'm not sure how that function works. I don't have hardware with me to check either.

Joel

promach, 2018-06-23:
@yunqu The image function at the bottom od the notebook does not display any picture. When I look up "mxp_filters_output.jpg" in the file directory, I find that the picture is all black, which means all zeroes in the numpy array. This basically means that the c coding has no effect on uint8_t *out_video

promach, 2018-06-25:
@vanjoe 

Since [mxp_filters.ipynb](https://gist.github.com/promach/24094b931979e4135c3c55813a000cf9) does not work accordingly with incorrect output image, I have created [mxp_data_sharing.ipynb](https://gist.github.com/promach/6db9343af8b25a007687f287b34cef6c) to verify the code at [VectorBlox MXP Programming Guide for Xilinx](http://vectorblox.github.io/mxp/mxp_guide_xilinx.html#data_sharing)

Could you advise if I have modified [makefile](https://paste.ubuntu.com/p/7y6RB63bCn/) incorrectly since I have an error related to the makefile in the ipynb file ?

promach, 2018-06-25:
@vanjoe @yunqu @PeterOgden @schelleg 

I have solved the previous makefile errors.

However, how should I modify [this call() function](https://github.com/VectorBlox/PYNQ/blob/vectorblox/python/pynq/iop/mxp_filter.py#L70-L85) for the data_sharing demo code ?

This code segment involves the use of **mmio** and some **offset** which I think user will not have extensive low-level hardware information about. Could you advise ?

vanjoe, 2018-06-25:
I think the function which you called `data_sharing` must be called `filter` like in the original example. 

Also I don't think the microblaze that the code is running on has the ability to print to stdout.

promach, 2018-06-26:
> I don't think the microblaze that the code is running on has the ability to print to stdout

@vanjoe 

In this case, could you advise how I should modify the memory region parameters in [lscript.ld](https://paste.ubuntu.com/p/RBvRntGCC4/) in order to allow printf to stdout ?

promach, 2018-07-01:
@vanjoe 

Even if I eventually find out the memory address of the MXP IP, I will still be stucked at figuring out the incoming command for MXP IP as in https://github.com/VectorBlox/PYNQ/blob/vectorblox/python/pynq/iop/mxp_filter.py#L79 

There are no obvious documentations that documents the link between MXP hardware IP core and the software command registers, right ?

Could you advise ?

schelleg, 2018-09-13:
I'll close here - please reopen on the VectorBlox repo if MXP-specific questions.  Otherwise, feel free to open a new issue.

closed 2018-09-13 by schelleg