Xilinx/PYNQ #668: PYNQ update issue

aniqurrehman, 2018-07-19:
while updating pynq i am getting this error.

fatal: unable to access 'https://github.com/Xilinx/PYNQ.git/': Could not resolve host: github.com
Command "git clone -q https://github.com/Xilinx/PYNQ.git /tmp/pip-dw4yjj8f-build" failed with error code 128 in None

i am using follwing command to update.
sudo pip3.6 install --upgrade git+https://github.com/Xilinx/PYNQ.git

aniqurrehman, 2018-07-19:
yes i am connecting my board to ethernet and i am also able to install other packages but reciveing error for this command.

aniqurrehman, 2018-07-19:
i am unable to install any package from git.

yunqu, 2018-07-19:
I think you must have installed git again on your board. The PYNQ image comes with git so you do not need to do that. I am not sure how to recover the original version but a simple solution could be re-image the sd card.

aniqurrehman, 2018-07-19:
no i haven't installed git again.
foolwing is the output for traceroute github.
[image]

yunqu, 2018-07-19:
Is it possible that you are behind a proxy?

aniqurrehman, 2018-07-20:
yes i am behind a proxy but i set the git proxy  and system proxy but still its not working 

[image]

aniqurrehman, 2018-07-20:
when i ping the proxy server it says network is unreachable.

yunqu, 2018-07-20:
did you try;

```python
export http_proxy=https://your_proxy:your_port
export https_proxy=http://your_proxy:your_port
```

aniqurrehman, 2018-07-21:
yes i tried that at that time i was behind proxy and it was not working now i left the proxy network and using another network and remove all proxies from github and system but now getting this error. i am stucked and don't know what to do with this . i've tried every method. but its not working.
[image]

drichmond, 2018-07-21:
Can you post the output of running `env`?

aniqurrehman, 2018-07-21:
BOARD=Pynq-Z1
XDG_SESSION_ID=c1
PYNQ_PYTHON=python3.6
SHELL=/bin/bash
TERM=vt220
HUSHLOGIN=FALSE
LC_ALL=en_US.UTF-8
USER=xilinx
RESIZED=1
MAIL=/var/mail/xilinx
PATH=/opt/sigrok/bin:/opt/sigrok/bin:/opt/microblazeel-xilinx-elf/bin:/opt/python3.6/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games
PWD=/home/xilinx
LANG=en_US.UTF-8
SHLVL=1
HOME=/home/xilinx
LOGNAME=xilinx
PYNQ_JUPYTER_NOTEBOOKS=/home/xilinx/jupyter_notebooks
XDG_RUNTIME_DIR=/run/user/1000
_=/usr/bin/env

drichmond, 2018-07-21:
Okay, nothing there looks worrisome. 

Are you able to ping github?

aniqurrehman, 2018-07-21:
yes i can ping github

aniqurrehman, 2018-07-22:
i have removed git and reinstall it but it didin't worked.

drichmond, 2018-07-22:
What about `sudo env`?

aniqurrehman, 2018-07-24:
i reinstall the image now the issue is resolved.

closed 2018-07-24 by aniqurrehman