Xilinx/PYNQ #677: Default SIGINT handler overridden by pynq package?

AndreasLoow, 2018-08-14:
I don't know if running Python programs based on the pynq package outside Jupyter is supported, but it seems to work. However, it seems that loading any pynq package (e.g. `from pynq import MMIO`) overrides the default SIGINT handler because pressing C-c when waiting for `read` in e.g.

```
import sys
from pynq import MMIO

try:
    c = sys.stdin.read(4)
finally:
    print("!!!")
```

results in the program saying "Application Terminated by User" and then aborting, never reaching `print("!!!")`; whereas the default handler throws `KeyboardInterrupt`, so that `print("!!!")` is reached.

I think some non-python code is overriding the handler, because `signal.getsignal(signal.SIGINT)` says that the default handler (`<built-in function default_int_handler>`) is used -- but clearly it isn't the default that's actually used! Or it might be something else happening? Anyway, I think this is a bug because loading pynq packages shouldn't affect C-c handling.

PeterOgden, 2018-09-17:
I've been looking into this. It appears to be a symptom of a package we depend on as PYNQ doesn't do anything explicit with SIGINT. I've not narrowed it down any further yet. I'll update when I find something.

AndreasLoow, 2018-09-20:
Do you mean the 2.3 version available here http://www.pynq.io/board.html?

PeterOgden, 2018-09-20:
Yes

AndreasLoow, 2018-09-27:
Thanks! The example I included in this issue seems to work with 2.3!

closed 2018-09-27 by AndreasLoow