Xilinx/PYNQ #737: Resolution is recognized wrong as 1080p

robotrovsky, 2018-09-28:
I always get a hdmi_in resolution of 1920x1080 independent of what the real resolution (GraphicsCard output) is. When i first hooked up the board it was displaying 720p just fine. I experimented with the resolutions and switched to 1080p from that time on it never again detected any lower resolution. I also installed a fresh pynq-image on the SD, but the behaviour doesnt change. 

1. What could be the problem?
2. Is it possible to pass a resolution to hdmi_in when calling configure()?

PeterOgden, 2018-09-28:
What do the frames coming out of the video pipeline look like? Are they scaled or is there a large amount of black space? We've had some experience of Intel integrated graphics cards outputting scaled resolutions even when Windows thinks it is a lower resolution and there is a setting buried deep in the Intel graphics control panel.

robotrovsky, 2018-09-28:
@PeterOgden thank you for your super fast support!! Actually the problem was not the pynq, but the monitor. Everything is working as expected even with 720p. 

But still if i have the mode set to 720p after
    hdmi_in.configure()
it reports
    print("%d x %d" %(hdmi_in.mode.width, hdmi_in.mode.height))
    1920 x 1080

But that should be no problem for now. So i will close this issue. Thank you very much again!!

closed 2018-09-28 by robotrovsky