Xilinx/PYNQ #738: AxiIIC usage issue

Swacheta, 2018-09-29:
We are trying to use the AxiIIC library but not able to progress. 
Can some please help. There is no example on how to use the AxiIIC.

```
> from pynq import Overlay
> overlay=Overlay("YaViCamera.bit)
> from pynq.lib.iic import AxiIIC
> i2c = overlay.ip_dict['axi_iic_0']
> i2cNew = AxiIIC(i2c)
```
ERROR message:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/xilinx/pynq/lib/iic.py", line 80, in __init__
    AxiIIC._initialise_lib()
  File "/home/xilinx/pynq/lib/iic.py", line 68, in _initialise_lib
    AxiIIC._lib = AxiIIC._ffi.dlopen(_lib_location)
  File "/usr/lib/python3/dist-packages/cffi/api.py", line 141, in dlopen
    lib, function_cache = _make_ffi_library(self, name, flags)
  File "/usr/lib/python3/dist-packages/cffi/api.py", line 802, in _make_ffi_library
    backendlib = _load_backend_lib(backend, libname, flags)
  File "/usr/lib/python3/dist-packages/cffi/api.py", line 797, in _load_backend_lib
    raise OSError(msg)
OSError: cannot load library '/usr/local/lib/python3.6/dist-packages/pynq/lib/libiic.so': /usr/local/lib/python3.6/dist-packages/pynq/lib/libiic.so: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called '/usr/local/lib/python3.6/dist-packages/pynq/lib/libiic.so'

sumilao, 2019-04-29:
You can use the img PYNQ 2.4，or copy the /usr/local/lib/python3.6/dist-packages/pynq/lib/libiic.so in img PYNQ 2.4 or the attechment to the same path in your rootfs.
[libiic.zip](https://github.com/Xilinx/PYNQ/files/3126279/libiic.zip)

giunatale, 2019-09-27:
Hi @Swacheta. Looks like your issue was answered elarlier.
However, even if you are not satisfied, I kindly need to ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase.

Closing this now, thanks in advance for your collaboration.

closed 2019-09-27 by giunatale