Xilinx/PYNQ #739: How encrypt a bitstream file?

ileop, 2018-10-10:
Hi, 
is it possible to encrypt a Bitstream file and still being able to use it as an overlay?
In the case, may you provide some hint on how to can do it, please?

Thank you

cathalmccabe, 2019-02-04:
Yes, this should be possible but it is largely orthogonal to PYNQ. Once you encrypt the bitstream, you should be able to download it with the PYNQ _Overlay_ class.

giunatale, 2019-09-27:
Hi @ileop, I see your question was already answered. If you are still not satisfied, I need to kindly ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase.

Closing this now, thanks in advance for your collaboration.

closed 2019-09-27 by giunatale