Xilinx/PYNQ #744: xilinx@192.168.2.99's password: Permission denied (publickey,password).  What is problem?  thanks

aysegulucar2016, 2018-10-19:
xilinx@192.168.2.99's password:
Permission denied (publickey,password).

What is problem?

thanks

Niaoutheron, 2018-10-24:
I had the same issue before while trying to connect from my browser. Did you configure your connection as static IP address?
I had the Permission Denied issue as well and had to switch to Google Chrome to allow my browser to access the card, give it a try!

aysegulucar2016, 2018-10-24:
Many thanks for reply 
You means "switch to Google Chrome to allow my browser to access the card,"

How can I do it?  I cant understand. Coud you please explain it? 

In addition. 
Now, again I did it. I shared wired my compter internet. I see yellow led relating to internet on  the board. But
That is same error. 
sudo gedit /etc/network/interfaces
auto lo eth0
iface lo inet loopback
iface eth0 inet static
        address 192.168.2.1
        netmask 255.255.255.0

aucar@aucar:~$ ifconfig
eth0      Link encap:Ethernet  HWaddr 20:47:47:78:09:d7  
          inet addr:10.42.0.1  Bcast:10.42.0.255  Mask:255.255.255.0
          inet6 addr: fe80::da84:85ea:f69:271a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:165 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:22070 (22.0 KB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:537 errors:0 dropped:0 overruns:0 frame:0
          TX packets:537 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:51083 (51.0 KB)  TX bytes:51083 (51.0 KB)

wlan0     Link encap:Ethernet  HWaddr 00:db:df:5d:6a:d3  
          inet addr:10.92.40.83  Bcast:10.92.255.255  Mask:255.255.0.0
          inet6 addr: fe80::c91e:7856:2102:bd6f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:60166 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4525 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:12771644 (12.7 MB)  TX bytes:627719 (627.7 KB)

aucar@aucar:~$ sudo ifconfig eth0 192.168.2.99
aucar@aucar:~$ ifconfig
eth0      Link encap:Ethernet  HWaddr 20:47:47:78:09:d7  
          inet addr:192.168.2.99  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::da84:85ea:f69:271a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:177 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:24677 (24.6 KB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:577 errors:0 dropped:0 overruns:0 frame:0
          TX packets:577 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:54232 (54.2 KB)  TX bytes:54232 (54.2 KB)

wlan0     Link encap:Ethernet  HWaddr 00:db:df:5d:6a:d3  
          inet addr:10.92.40.83  Bcast:10.92.255.255  Mask:255.255.0.0
          inet6 addr: fe80::c91e:7856:2102:bd6f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:61292 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4673 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:12879782 (12.8 MB)  TX bytes:644748 (644.7 KB)

aucar@aucar:~$ ssh xilinx@192.168.2.99
xilinx@192.168.2.99's password: 
Permission denied, please try again.
xilinx@192.168.2.99's password:

Niaoutheron, 2018-10-24:
I installed Chrome to access http://192.168.2.99:9090 since my other internet browser could not do that. I did not use the UART Linux debug to access the card but only via Jupyter Notebook and Samba

aysegulucar2016, 2018-10-24:
Many thanks for answer. I am using chrome.
My error is ..
This site can’t be reached
192.168.2.99 took too long to respond.
Try:

Checking the connection
Checking the proxy and the firewall

Nowö How I can access by means of Jupyter Notebook and Samba.

Niaoutheron, 2018-10-24:
Did you follow the steps in the tutorial for connecting your board? https://pynq.readthedocs.io/en/v2.0/index.html

aysegulucar2016, 2018-10-24:
Yes. I think so. I cant open browser also http://192.168.2.99:9090

yunqu, 2018-10-24:
You probably only need 1 internet connection. If you have multiple network connections that 192.168.2.99 can lead to some other hosts instead of the board. I would just make the static wired connection the only connection on my PC side. Also make sure to change the static IP address on your PC side.

aysegulucar2016, 2018-10-24:
After ping 192.168.2.99, I see yellow led on internet on board.  Many thanks i will check again static ip.

yunqu, 2018-10-24:
That yellow LED is flashing whenever there is some activity on the ethernet port. It does not mean your connection is properly set up...

closed 2018-10-24 by yunqu

gn0st1c, 2018-10-26:
@aysegulucar2016 : birden fazla interface'den ve ayni network'de olmadiginiz icin route'da problem olabilir.

aysegulucar2016, 2018-10-26:
Thanks for answer. I shared my wireless laptop connection. Then sudo ifconfig eth0 192.168.2.99 .
I used putty even. I cant solve the problem. I dont know  "birden fazla interface'de " problem. 
I gave up.

gn0st1c, 2018-10-26:
@aysegulucar2016:
1) if you connect your pynq directly to your laptop and execute "sudo ifconfig eth0 192.168.2.99", both pynq and your laptop would have the same ip. don't do that.

assuming your `/etc/network/interfaces` file looks something like this;
```
auto lo eth0
iface lo inet loopback
iface eth0 inet static
        address 192.168.2.1
        netmask 255.255.255.0
```

so when you `ifconfig eth0` you should see inet addr as 192.168.2.1;
```
...
eth0 Link encap:Ethernet HWaddr 00:00:00:00:00:00
inet addr:192.168.2.1 Bcast:192.168.2.255 Mask:255.255.255.0
...
```

if your inet addr is different than the one you set in your interfaces file (like the one you've pasted above), try the following commands (or restart networking daemon)
```
sudo ifconfig eth0 down
sudo ifconfig eth0 up
```

basically, your pynq has a static ip of 192.168.2.99, so your computer should be 192.168.2.X (where x anything but 99)

2) if you connect both your pynq and your laptop to the same network (where there is a DHCP server is running), your pynq will get a different ip (not 192.168.2.99). `ping pynq` should reveal the ip. (hint: avahi/zeroconf)

3) and finally, if you're still having trouble, just copy/paste the outputs of the following commands;
```
uname -a
more /etc/network/interfaces
route -n
```

aysegulucar2016, 2018-10-27:
Thanks for answer. I have still same problem.

aucar@aucar:~$ uname -a
Linux aucar 4.4.0-137-generic #163-Ubuntu SMP Mon Sep 24 13:14:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
aucar@aucar:~$ more /etc/network/interfaces
auto lo eth0
iface lo inet loopback
iface eth0 inet static
        address 192.168.2.1
        netmask 255.255.255.0
aucar@aucar:~$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG    600    0        0 wlan0
10.42.0.0       0.0.0.0         255.255.255.0   U     100    0        0 eth0
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 wlan0
192.168.1.0     0.0.0.0         255.255.255.0   U     600    0        0 wlan0
[image]

gn0st1c, 2018-10-27:
@aysegulucar2016: it seems your eth0 is managed by NetworkManager, so the interfaces file is ignored (thus your settings)

[more on NetworkManager](https://wiki.debian.org/NetworkManager#doc)

you can use your NetworkManager to set the static ip (IPv4 Settings) or force the NetworkManager to not manage your Wired Connection. (again refer the NetworkManager doc above)

aysegulucar2016, 2018-11-01:
Many thanks for answer. I still try it.

aysegulucar2016, 2018-11-06:
Finally I solved problem. I installed old image v2.1.  Many thanks to everyone.