Xilinx/PYNQ #750: the pynq dma can't support Single channel mode

hisrg, 2018-11-05:
if i set DMA as below picture. it can't work and find a error.
[image]
[image]
could you tell me how to solve it?

PeterOgden, 2018-11-05:
The dma object appears to have the wrong driver allocated to it. Can you post the full notebook so I can have a better look?

hisrg, 2018-11-06:
[image]
[image]
[image]

note: the bit file has changed so that report error as above when i reply this issue, you can use pynq tutorial2's bd file to generate bit and tcl file to overlay in this notebook, but it needs you setup the DMA  config as 1st figure. thanks.

PeterOgden, 2018-11-06:
The `bindto` statement in your ConstantMultiplyDriver is wrong - it's currently binding to the DMA engine which is not what you want. Given that you are not using that IP any longer I would just remove the driver and restart the notebook.

hisrg, 2018-11-23:
[image]
 how to write this DMA drvier as above bd? which produce some data from PL(1),and then writes in FIFO(2), Finally it will display in Jupyter notebook.  i think it trans data from PL to PS using singal DMA channel and interrupt.

yunqu, 2019-01-18:
There are many examples online. For example:
https://github.com/Xilinx/PYNQ-DL/blob/master/notebooks/resize.ipynb
https://github.com/yunqu/PYNQ-derivative-overlays/tree/master/sdx_platform#porting-onto-pynq

By the way, your S_AXIS should not be an input port, it should connect to another DMA back to PS, or another channel of your block 2. That way you have a complete loop.

giunatale, 2019-09-27:
Hi @hisrg, I see your question has been already extensively answered. If you are still not satisfied, I need to kindly ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase.

Closing this now, thanks in advance for your collaboration.

closed 2019-09-27 by giunatale