Xilinx/PYNQ #754: sdbuild flow dependency on sdx unclear

hzappen, 2018-11-09:
I ran into problems during image building, because the Makefile checks if 'sdx' is present. Before, i ran the setup_host.sh script successfully. The documentation and the readme file in the sdbuild subdirectory do not mention this dependency explicitely. 

What am i supposed to install here? I could only find out that sdx somehow belongs to the Xilinx SDAccel application? Is this really neccessary to build a basic image for a custom board, because afaik this software is only available as a paid license version. In version 2.2 of PYNQ is was able to build an image without installing this software.

Thanks,
Hendrik

xupsh, 2018-12-08:
sdx mean [SDSoC](https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/sdx-development-environments.html) here. Though it's not total free, you can have a trial in www.xilinx.com/getlicense

closed 2018-12-13 by yunqu

riple, 2019-03-02:
To build 2.4 of PYNQ, you need 2018.3 version of Vivado\SDK\SDx\Petalinux, of which the first 3 is involved in SDSoC installation. So it is better to download a single installer of SDSoC.

yunqu, 2019-03-02:
Maybe we should mention that even SDSoC is not required now; Vivado is required to build z1 and z2 bsp's. That is perhaps all you need. 

The image building for non-z1/z2 boards only requires petalinux.
The image building for z1/z2 boards needs petalinux + vivado.