Xilinx/PYNQ #755: PYNQ for ZC706

LovExtreme, 2018-11-13:
Hi,
Recently, I've read in the manual that the PYNQ can be retargeted into another board. So I try to give it a shot with my ZC706. But when I came to the build part, The script stopped with and error:

scripts/Makefile.build:59: board/zc706/Makefile: No such file or directory
make[2]: *** No rule to make target 'board/zc706/Makefile'.  Stop.
Makefile:1210: recipe for target 'board/zc706' failed
make[1]: *** [board/zc706] Error 2
make[1]: *** Waiting for unfinished jobs....

Then, I noticed that they need a config for ZC706 located in the ./boot_configs right?  And when I took a look into the folder of ./PYNQ-Z1-defconfig there some sort of files like u-boot.config, kernel.config, ps7_config.tcl, pynq.dtsi, Could you tell me where I can get these files. Currently I'm using the BSP from the Xilinx page:

https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-design-tools.html

Thank you

yunqu, 2018-11-14:
Hi, I think you should follow the guideline https://github.com/Xilinx/PYNQ/tree/master/sdbuild#porting-to-a-new-board

Nothing has to be touched for the boot configs (unless you know what you are doing). The key step is to prepare the board folder; in general the files in sdbuild folder should not be manually changed.

LovExtreme, 2018-11-21:
If anybody struggles with the same problem with compiling the PYNQ for ZC706. I suggest using the newest version of PYNQ v2.3 and Xilinx tool packs 2018.x . Then use BSP for ZC706 version 2018.x from Xilinx and the prebuilt image from `https://www.xilinx.com/member/forms/download/xef.html?filename=pynq_rootfs_arm_v2.3.zip`
The script from ./sdbuild/scripts/image_from_prebuilt.sh will take both of them and save a lot of time for you.

closed 2018-11-21 by LovExtreme

yuginhc, 2018-12-05:
> If anybody struggles with the same problem with compiling the PYNQ for ZC706. I suggest using the newest version of PYNQ v2.3 and Xilinx tool packs 2018.x . Then use BSP for ZC706 version 2018.x [...]

Can you share the Pynq image for ZC706 or manual how to make it?

LovExtreme, 2018-12-06:
@yuginhc 
1. First you need the BSP file for your board. You can download it from [here](https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-design-tools.html)
2. Download the pre-built image for arm here https://www.xilinx.com/member/forms/download/xef.html?filename=pynq_rootfs_arm_v2.3.zip
2. Clone the PYNQ source v2.3 from Github.
3. Go to the <where-you-clone-pynq>/sdbuild/ there is a .sh script name scripts/image_from_prebuilt.sh
Here is its description:
```
image_from_prebuilt.sh board bsp_file arch prebuilt
Script for generating a board-specific image from a board agnostic prebuilt image

board    : The name of the board
bsp_file : The BSP file to use
arch     : The architecture of the system - either arm or aarch64
prebuilt : The image to use as a base
```
Now board is "zc706". bsp_file is where you put the BSP file. arch will be "arm" and finally is the image pre-built you have downloaded.

amiinous, 2019-05-15:
Hey @LovExtreme 
I walked through the instructions, but after deploying the image to my ZC706 board, I cannot connect to it via ethernet (just UART). Performing an ifconfig just shows the "lo". It is worth to mention that ifconfig -a shows the eth0 but no still no inet is visible.

frankieliu, 2019-05-15:
@amiinous let me know if you found a workaround, thanks!

cathalmccabe, 2019-05-15:
@frankieliu We have a support forum here: https://discuss.pynq.io You may want to open a post.

shravan-red, 2019-08-02:
Terminal returned the sam thing on executing the below command :
./scripts/image_from_prebuilt.sh zedboard /home/abcd/avnet-digilent-zedboard-v2018.2-final.bsp arm /home/abcd/pynq_z1_v2.3.img /home/abcd/PYNQ/boards/Zedboard/base/base.bit
says

> @yuginhc
> 
>     1. First you need the BSP file for your board. You can download it from [here](https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-design-tools.html)
> 
>     2. Download the pre-built image for arm here https://www.xilinx.com/member/forms/download/xef.html?filename=pynq_rootfs_arm_v2.3.zip
> 
>     3. Clone the PYNQ source v2.3 from Github.
> 
>     4. Go to the /sdbuild/ there is a .sh script name scripts/image_from_prebuilt.sh
>        Here is its description:
> 
> 
> ```
> image_from_prebuilt.sh board bsp_file arch prebuilt
> Script for generating a board-specific image from a board agnostic prebuilt image
> 
> board    : The name of the board
> bsp_file : The BSP file to use
> arch     : The architecture of the system - either arm or aarch64
> prebuilt : The image to use as a base
> ```
> 
> Now board is "zc706". bsp_file is where you put the BSP file. arch will be "arm" and finally is the image pre-built you have downloaded.

jcooklis, 2019-11-20:
I just tried to do this for PYNQ 2.5 using version 2019.1 of the tools, and I got the same results as amiinous (no Ethernet):

image_from_prebuilt.sh zc706 xilinx-zc706-v2019.1-final.bsp arm bionic.arm.2.5.img

Did you find a way to get the network up and running with this method?

giunatale, 2019-11-20:
@jcooklis this is a closed issue so I would kindly ask you to redirect your question to our support forum https://discuss.pynq.io
We reply fairly regularly there.

mmdsaifudeen, 2019-12-09:
I am building PYNQ System for ZC706.And got successful boot.But display did not come.It is showing ""   No Screen found(EE) ""  on doing xinit
Any one please do help...

giunatale, 2019-12-09:
@mmdsaifudeen as I already mentioend, this is a closed issue. We have a dedicated forum for this sort of questions https://discuss.pynq.io where our team and the rest of the community actively provide support for issues like this one.
Please, post your question there and I also suggest you detail as much as possible the steps taken and the errors encountered, as it's really hard to understand what's happening by your comment here.

Thanks for your collaboration.

mmdsaifudeen, 2019-12-09:
yes ,I will do there..thanks