Xilinx/PYNQ #756: Can't build PYNQ images from source

Olangu, 2018-11-13:
I'm trying to build the PYNQ images from this repository but they fail with the error below.

```
make BOARDDIR=/home/angu/PYNQ/myboards /opt/qemu/bin/qemu-arm-static -version | fgrep 2.8.0
qemu-arm version 2.8.0
which vivado | fgrep 2018.2
/opt/Xilinx/Vivado/2018.2/bin/vivado
which sdx | fgrep 2018.2
/opt/Xilinx/SDx/2018.2/bin/sdx
which petalinux-config | fgrep 2018.2
/opt/Xilinx/PetaLinux/2018.2/tools/common/petalinux/bin/petalinux-config
which arm-linux-gnueabihf-gcc
/opt/Xilinx/SDK/2018.2/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin/arm-linux-gnueabihf-gcc
which microblaze-xilinx-elf-gcc
/opt/Xilinx/SDK/2018.2/gnu/microblaze/lin/bin/microblaze-xilinx-elf-gcc
which ct-ng
/opt/crosstool-ng/bin/ct-ng
which python | fgrep /usr/bin/python
/usr/bin/python
sudo -n mount > /dev/null
bash /home/angu/PYNQ/sdbuild/scripts/check_Env.sh
bash /home/angu/PYNQ/sdbuild/scripts/check_mounts.sh
PYNQ_BOARDNAME=Pynq-Z2 petalinux-build -c bootloader -p /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project
[INFO] building bootloader
[INFO] sourcing bitbake
INFO: bitbake virtual/fsbl
Loading cache: 100% |#################################################| Time: 0:00:00
Loaded 3423 entries from dependency cache.
Parsing recipes: 100% |###############################################| Time: 0:00:08
Parsing of 2552 .bb files complete (2517 cached, 35 parsed). 3425 targets, 148 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
Initialising tasks: 100% |############################################| Time: 0:00:01
Checking sstate mirror object availability: 100% |####################| Time: 0:00:05
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: fsbl-2018.2+gitAUTOINC+6e82c0183b-r0 do_configure: Function failed: do_configure (log file is located at /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/fsbl/2018.2+gitAUTOINC+6e82c0183b-r0/temp/log.do_configure.9189)
ERROR: Logfile of failure stored in: /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/fsbl/2018.2+gitAUTOINC+6e82c0183b-r0/temp/log.do_configure.9189
Log data follows:
| DEBUG: Executing shell function do_configure
| MISC_ARG is  -yamlconf /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/fsbl/2018.2+gitAUTOINC+6e82c0183b-r0/fsbl.yaml
| APP_ARG is  -app "Zynq FSBL"
| cmd is: xsct /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/fsbl/2018.2+gitAUTOINC+6e82c0183b-r0/app.tcl -ws /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/../components/plnx_workspace/fsbl -pname fsbl -rp /opt/Xilinx/PetaLinux/2018.2/tools/hsm/data/embeddedsw -processor ps7_cortexa9_0 -hdf /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/deploy/images/plnx-zynq7/Xilinx-plnx-zynq7.hdf -arch 32  -app "Zynq FSBL"  -yamlconf /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/fsbl/2018.2+gitAUTOINC+6e82c0183b-r0/fsbl.yaml
| Display is :177
| Starting xsdk. This could take few seconds... ERROR: [#UNDEF] Java VM required by SDK is not found in the Installation
| XSCTHELPER INFO: Empty WorkSpace
| Starting xsdk. This could take few seconds... ERROR: [#UNDEF] Java VM required by SDK is not found in the Installation
| timeout while establishing a connection with SDK
|     while executing
| "error "timeout while establishing a connection with SDK""
|     (procedure "getsdkchan" line 111)
|     invoked from within
| "getsdkchan"
|     (procedure "::sdk::set_user_repo_path_sdk" line 16)
|     invoked from within
| "::sdk::set_user_repo_path_sdk $params(set)"
|     (procedure "repo" line 27)
|     invoked from within
| "repo -set $path"
|     invoked from within
| "if { $params(ws) ne "" } {
| 	#Local Work Space available
| 	setws $params(ws)
| 	if { [catch {importprojects $params(ws)} result] } {
| 		puts "XSCTHELPER IN..."
|     (file "/home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/fsbl/2018.2+gitAUTOINC+6e82c0183b-r0/app.tcl" line 120)
| WARNING: /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/fsbl/2018.2+gitAUTOINC+6e82c0183b-r0/temp/run.do_configure.11768:1 exit 1 from 'eval xsct /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/fsbl/2018.2+gitAUTOINC+6e82c0183b-r0/app.tcl -ws /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/../components/plnx_workspace/fsbl -pname fsbl -rp /opt/Xilinx/PetaLinux/2018.2/tools/hsm/data/embeddedsw -processor ps7_cortexa9_0 -hdf /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/deploy/images/plnx-zynq7/Xilinx-plnx-zynq7.hdf -arch 32 ${APP_ARG} ${MISC_ARG}'
| ERROR: Function failed: do_configure (log file is located at /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/fsbl/2018.2+gitAUTOINC+6e82c0183b-r0/temp/log.do_configure.11768)
ERROR: Task (/opt/Xilinx/PetaLinux/2018.2/components/yocto/source/arm/layers/meta-xilinx-tools/recipes-bsp/fsbl/fsbl_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 617 tasks of which 609 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /opt/Xilinx/PetaLinux/2018.2/components/yocto/source/arm/layers/meta-xilinx-tools/recipes-bsp/fsbl/fsbl_git.bb:do_configure
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
ERROR: Failed to build bootloader
Makefile:296: recipe for target '/home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/images/linux/zynq_fsbl.elf' failed
make: *** [/home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/images/linux/zynq_fsbl.elf] Error 255
```

What is wrong? It seem to have unset the Java VM, but why?

Running the failing command manually works fine.

```
$ xsct /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/fsbl/2018.2+gitAUTOINC+6e82c0183b-r0/app.tcl -ws /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/../components/plnx_workspace/fsbl -pname fsbl -rp /opt/Xilinx/PetaLinux/2018.2/tools/hsm/data/embeddedsw -processor ps7_cortexa9_0 -hdf /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/deploy/images/plnx-zynq7/Xilinx-plnx-zynq7.hdf -arch 32  -app "Zynq FSBL"  -yamlconf /home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/fsbl/2018.2+gitAUTOINC+6e82c0183b-r0/fsbl.yaml
rlwrap: warning: your $TERM is 'xterm-256color' but rlwrap couldn't find it in the terminfo database. Expect some problems.
Starting xsdk. This could take few seconds... done                                   
No project(s) found in: '/home/angu/PYNQ/sdbuild/build/Pynq-Z2/petalinux_project/components/plnx_workspace/fsbl'
XSCTHELPER INFO: Empty WorkSpace
INFO: [Hsi 55-1698] elapsed time for repository loading 0 seconds                    
INFO: create app using fsbl_hwproj project                                           
WARNING: [Hsi 61-9] Current Software design may not be compatible with "zynq_fsbl" app. Tool is ignoring the MSS file specified in the app directory
generate_target: Time (s): cpu = 00:00:06 ; elapsed = 00:00:07 . Memory (MB): peak = 494.859 ; gain = 30.133 ; free physical = 4627 ; free virtual = 21427
hsi::generate_app: Time (s): cpu = 00:00:07 ; elapsed = 00:00:07 . Memory (MB): peak = 494.867 ; gain = 31.070 ; free physical = 4630 ; free virtual = 21426
Application project 'fsbl' created successfully.       
```

Please help!

yunqu, 2018-11-13:
Did you first try copying pynq-z2 into /myboards and build?

Olangu, 2018-11-13:
> 
> 
> Did you first try copying pynq-z2 into /myboards and build?

Yes of course!

yunqu, 2018-11-13:
The javaVM problem is very strange; not sure if your petalinux is installed correctly?

Olangu, 2018-11-14:
Reinstallation of PetaLinux did in fact solve this problem.
The root cause might have been that PetaLinux was moved after installation.

Maybe there should be some sanity check in the PetaLinux installation.

Thank you!

closed 2018-11-14 by yunqu