Xilinx/PYNQ #761: Some questions about using Pynq xlnk.

clancylea, 2018-11-15:
I'm a pynq user, i have some questions when using pynq .
I allocated memory by pynq xlnk() like Figure(1), and send the physical address to the linux application, which use mmap() function to remap the physical address. But the result was wrong.
Before this, i used Zedborad with half memory for Linux and half for PL(Because Zedboard only has 512MB memory, so memory from 0x1000000 to 0x1FFFFFFF are used by PL), using mmap() to remap physical address above 0x10000000,
 and the result was right.
I didn't get the key to find where's the problem, can you give me some advice? Thanks!

[image]

cathalmccabe, 2019-02-04:
What are you trying to do? Xlnk allocates contiguous memory in the Linux memory system, and gives the virtual address and the corresponding physical address. Why are you trying to remap the memory from Linux? You can already access this memory from Linux using the virtual address.

giunatale, 2019-09-27:
Hi @clancylea, I see your question was already answered. If you are still not satisfied, I need to kindly ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase.

Closing this now, thanks in advance for your collaboration.

closed 2019-09-27 by giunatale