Xilinx/PYNQ #766: Asynchronous Pmod_IO or Arduino_IO?

VicerExciser, 2018-11-30:
I have a coroutine (implemented using the `asyncio` library) that polls temperature values in a background-ish way from sensors on a 1-Wire bus. If a read temperature is too high or too low, the coroutine will open or close a pilot gas valve that I have connected via PMOD.

However, whenever Pmod_IO's write() function is called for the device, it incurs a RuntimeError from pmod_devmode.py like this:
`File "/opt/python3.6/lib/python3.6/site-packages/pynq/lib/pmod/pmod_io.py", line 131, in write
    PMOD_DIO_DATA_OFFSET)
  File "/opt/python3.6/lib/python3.6/site-packages/pynq/lib/pmod/pmod_devmode.py", line 263, in read_cmd
    raise RuntimeError("PmodDevMode read_cmd() not acknowledged.")
RuntimeError: PmodDevMode read_cmd() not acknowledged.`

The same occurs with Arduino_IO. Is it possible to use these I/O modules with `asyncio`?

VicerExciser, 2018-12-01:
So I have now confirmed that Pmod_IO will work when instantiating BaseOverlay using "base.bit" and my custom overlay (an extended base design), and will also work for "base.bit" after changing Clocks.fclk0_mhz to 33.33333.

However, Pmod_IO breaks for my custom overlay only after modifying fclk0's frequency, even after setting it back to 100.0 MHz again! It may be something within the custom IP that I've added, but I can't imagine why changing a function clock frequency would raise an error when PmodDevMode read_cmd()? Is this indicating that the cmd_mailbox is never idle??

yunqu, 2018-12-13:
When you change the clock, does it mean you keep the overlay on PL and only change the clock frequency? That might be disruptive to the microblaze processors?

giunatale, 2019-09-27:
Hi @VicerExciser, I need to kindly ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase.

Closing this now, thanks in advance for your collaboration.

closed 2019-09-27 by giunatale