Xilinx/PYNQ #771: update problem of pynq

aysegulucar2016, 2018-12-08:
ilinx@pynq:~$ sudo apt-get update
[sudo] password for xilinx: 
Err:1 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
  Temporary failure resolving 'ports.ubuntu.com'
Reading package lists... Done
W: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/bionic/InRelease  Temporary failure resolving 'ports.ubuntu.com'
W: Some index files failed to download. They have been ignored, or old ones used instead.

xupsh, 2018-12-08:
Seems to be network error. Have a try to `ping ports.ubuntu.com` to see if the network is ok.

hustjinghu, 2019-01-01:
How to discover the pynq version in the linux shell?

hustjinghu, 2019-01-01:
Dear xupsh , i have ping ports.ubuntu.com,but the following results shows. However,i ping 192.168.2.99, it shows my PC can connect to PYNQ-Z2 board, my PC can connecte to internet.
[image]

closed 2019-01-18 by yunqu