Xilinx/PYNQ #780: some questions about network

wangjiefjj, 2019-01-09:
I have set PC a static IP address  already,  and the PC can ping the board, but the board cannot ping the PC.
what's more, when I use "sudo pip3 install git+https:XXX" on the board, it was successed, but when I ping www.XXX.com , it returns " failed: Temporary failure in name resolution".  what is the reasons  for that?

schelleg, 2019-01-10:
Hi @wangjiefjj - your board doesn't have internet connectivity when using a static IP address to communicate between the board and a host machine.  If you need internet connectivity for the board, you can plug the board directly into a router, use a wifi dongle + wifi connectivity, or try port forwarding on our host machine.

closed 2019-01-18 by yunqu

xmy0053, 2020-04-20:
The problem is caused by the static IP 192.168.2.99;
If u want to directly connect the board to your PC and make your  board have access to the Internet, you should set the board static IP between 192.168.137.2~192.168.137.254. Set your host PC Ethernet IP 192.168.137.1 and bridge your Wifi to Ethernet. Static IP 192.168.2.99 causes a lot of problems especially in the university since there are so many device within this range.