Xilinx/PYNQ #787: Not able to access internet behind proxy server

tanishqjasoria, 2019-01-17:
I have tried the to configure the by using
set http_proxy=my_http_proxy:8080
set https_proxy=my_https_proxy:808
But it didn't work!

yunqu, 2019-01-18:
Hi, make sure your http proxy server address is correct. For me, I use my_http_proxy for both http and https. Also, make sure port number is correct.

tanishqjasoria, 2019-01-18:
i have first tried with
set http_proxy=172.16.2.30:8080
set https_proxy=172.16.2.30:8080

then with
set http_proxy=http://172.16.2.30:8080
set https_proxy=http://172.16.2.30:8080

then with
set http_proxy=http://172.16.2.30:8080
set https_proxy=https://172.16.2.30:8080

none of them worked!

cathalmccabe, 2019-02-04:
Did you check the environment variable was set?
The default shell is bash, so you should use 'export'
export http_proxy=http://x.x.x.x:8080/

giunatale, 2019-09-27:
Hi @tanishqjasoria, I see your question was already answered. If you are still not satisfied, I need to kindly ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase.

Closing this now, thanks in advance for your collaboration.

closed 2019-09-27 by giunatale