Xilinx/PYNQ #817: [PYNQ-Z1 v2.3] gdb broken after apt upgrade

umarcor, 2019-02-17:
With a fresh [PYNQ-Z1 v2.3 SDCard image](http://files.digilent.com/Products/PYNQ/pynq_z1_v2.3.zip), `gdb /bin/ls` works as expected:

``` bash
xilinx@pynq:~$ gdb /bin/ls
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /bin/ls...(no debugging symbols found)...done.
(gdb) r
Starting program: /bin/ls
Cannot parse expression `.L1207 4@r4'.
warning: Probes-based dynamic linker interface failed.
Reverting to original interface.

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
jupyter_notebooks  pynq  REVISION
[Inferior 1 (process 28611) exited normally]
(gdb) q
```

However, after executing `apt update` and `apt upgrade`, it no longer works:

``` bash
xilinx@pynq:~$ gdb /bin/ls
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /bin/ls...(no debugging symbols found)...done.
(gdb) r
Starting program: /bin/ls
warning: the debug information found in "/usr/lib/debug//lib/arm-linux-gnueabihf/ld-2.27.so" does not match "/lib/ld-linux-armhf.so.3" (CRC mismatch).

Program received signal SIGSEGV, Segmentation fault.
0xb6fd8f4e in ?? () from /lib/ld-linux-armhf.so.3
(gdb) q
```

PeterOgden, 2019-02-28:
For licensing reasons we rebuild all of the Ubuntu packages we ship as part of the image, this means that things like debuginfo will not be consistent with those in the main Ubuntu repositories. Easiest solution may be to try and reinstall all of the packages from the Ubuntu repo. I'm currently experimenting with that approach to see if it is viable.

PeterOgden, 2019-03-01:
I just ran the following set of commands as root

```
dpkg --get-selections | cut -f1 > packages
for f in `cat packages`; do apt-get install --reinstall $f; done
```

after running `apt-get update` and `apt-get upgrade`. This will reset some of the configurations as well so that, for example, the serial console no longer logs in automatically but it resolved the gdb issue.

giunatale, 2019-09-27:
Hi @umarcor, I see your question was already answered. If you are still not satisfied, I need to kindly ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase.

Closing this now, thanks in advance for your collaboration.

closed 2019-09-27 by giunatale

umarcor, 2019-09-27:
@giunatale, I did not ask any question at all. I did report a bug: ubuntu SD images for PYNQ are fundamentally broken. It is neither a regular Ubuntu nor a custom distribution. It is not documented which features/tools should work as regular packages or otherwise.

@PeterOgden, mentioned that you rebuild all the packages *for licensing issues*. That's a comment or a fact, not an explanation. Which are those issues that require a free/libre and open source tool to be recompiled? Are non-free resources being built into these packages?

@PeterOgden also mentioned that he was experimenting with installing packages from the Ubuntu repo. If https://github.com/Xilinx/PYNQ/issues/817#issuecomment-468607306 is the answer, why is not contributed to master? If it is not, what needs to be done to preserve the configuration or, at least, to know what is changed?

> That's the dedicated place to receive support from the community.
> The issue tracker here is to report bugs in the codebase.

I'm not asking for support, I'm asking for `apt update && apt upgrade` to work straight-away without *dirty* fixes. I believe that no one in the community can help fix an issue related to some non-public pre-built content distributed from Xilinx servers.

If you don't want to fix this because you don't expect users to update packages at all, but rather put their confidence on your *static* distribution of closed-origin binaries, you are of course entitled to mark this as `wontfix` and leave it closed.

giunatale, 2019-09-30:
Hi @umarcor, you are right, yours was not a question. I intended "your report was addressed", and I apologize for the misunderstanding. Thruth is though, Peter kind of gave a definitve answer already in his reply.

The problem is that for licensing reasons, this is an unfixable issue. Our images are not broken, they are simply frozen. As a small team, it's impossible for us to mantain and update **every** new version of **every** package, so we do not provide a ppa repository for updating your package. We instead set the ubuntu ones in case users might want to use this feature right before packaging the image. However, packages installed in the image are built and distributed by us, so they are **not** and cannot be the same packages Ubuntu distributes, again for licensing reasons. Simply put, this means that we, as a small team, distribute the image as is, with all the packages at the given version, and cannot guarantee that updating them through `apt` will work. I hope this clears your doubts.

I'll mark this issue as `wontfix` as you suggest, and leave it closed. Thanks again for your contribution!

umarcor, 2019-09-30:
Hi @giunatale! My most sincere gratitude for taking your time to clarify. And, please, accept my apologies if my previous message sounded *too overexcited*.

It'd be awesome if you could add some comment to the README or anywhere else to prevent other users from hitting this issue. E.g.:

> Our images are *frozen*, because it is not possible for us, as a small team, to maintain every new version of every package in a *ppa* repository. We set the ubuntu ones in case users might want to use this feature **right before packaging the image**.
> 
> Users that download pre-built SD images should avoid `apt upgrade`. See [Xilinx/PYNQ#817](https://github.com/Xilinx/PYNQ/issues/817) for details.