Xilinx/PYNQ #822: SD flow rebuild

schegde, 2019-02-25:
What is the recommended strategy for fastest end-to-end build for making some modifications to PYNQ rootfs packages and testing on my own board(Zybo)? I have been able to build an sd card image for Zybo, that boots properly and lands into the Ubuntu userspace. For hardware I assume, I can directly drop the the .bit and .tcl from Vivado into the userspace, in the right location and use the PYNQ overlay management APIs for direct testing. For some modification in rootfs packages, like usbgadget PYNQ package(changing the default IP address for eg.), that gets added into image put on SD card., I would have to do a make BOARDDIR={Zybo board dir} and that takes a long time(~1 hr) to produce the .img ,which i can `dd` into the SD card. Is there a fast "filesystem" only recompile option, which doesn't rebuild the entire ubuntu filesystem from scratch, but simply adds it and provides a .img? I see a make sysroot option in documentation, is it for this purpose?

PeterOgden, 2019-02-28:
The SD image build flow is designed such that it checkpoints at three stages and assuming that non of the dependencies on each stage have changed, re-running make will start from the last checkpoint. These points are:

 1. After the Ubuntu system is bootstrapped
 2. After all packages except PYNQ and other light-weight are installed
 3. After a board-independent version of PYNQ is installed.

If you are making minor modifications you can add the package you are modifying either to STAGE3_PACKAGES in ubuntu/bionic/${ARCH}/config or to STAGE4_PACKAGES in your board's spec file.

Note that our dependency tracking isn't smart enough to keep track of files inside a package so you may need to delete the image manually to allow make to rebuild - the final checkpoint is output/bionic.${ARCH}.${VER}.img.

My flow when creating a new package is to start by having it be the sole stage 3 or 4 package allowing for fast iterations from the previous checkpoint.

schegde, 2019-03-03:
Thanks for the elaboration. Yes, the dependency tracking is not able to track changes within packages , but deleting the final checkpoint image ,along with STAGE4_PACKAGES allows for a much faster iteration.

closed 2019-03-03 by schegde