Xilinx/PYNQ #824: PYNQ Loading fails

tusu12, 2019-02-25:
After powering up the PYNQ board , I am getting this weird message in the PUTTY terminal and the board  got hanged.

Any comments on this issue will be appreciated.

xilinx@pynq:~$ Unhandled fault: imprecise external abort (0x1406) at 0xb6d6e3f4
pgd = c0004000
[b6d6e3f4] *pgd=00000000
Internal error: Oops - BUG: 1406 [#1] PREEMPT SMP ARM
Modules linked in:
CPU: 0 PID: 756 Comm: jbd2/mmcblk0p2- Not tainted 4.14.0-xilinx-v2018.2 #1
Hardware name: Xilinx Zynq Platform
task: d75b2040 task.stack: d6dd8000
PC is at journal_clean_one_cp_list+0x94/0xa4
LR is at __jbd2_journal_remove_checkpoint+0x74/0xdc
pc : [<c029039c>]    lr : [<c028fd7c>]    psr: 20060013
sp : d6dd9e58  ip : 00000000  fp : d770cdf4
r10: c0a4a22c  r9 : d6dd8000  r8 : 00000000
r7 : d70ef840  r6 : d70ef940  r5 : d6dd8000  r4 : d70ef980
r3 : 000f4040  r2 : 00000000  r1 : 00000000  r0 : 00000001
Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 18c5387d  Table: 16de004a  DAC: 00000051
Process jbd2/mmcblk0p2- (pid: 756, stack limit = 0xd6dd8210)
Stack: (0xd6dd9e58 to 0xd6dda000)
9e40:

cathalmccabe, 2019-04-18:
Are you using USB or wall power? If power from USB is dipping during boot, it could be causing something to fail. If you can, use wall power, or even a USB plug with higher power output. You may need to reflash your sd card first

giunatale, 2019-09-27:
Hi @tusu12, I see your question was already answered. If you are still not satisfied, I need to kindly ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase.

Closing this now, thanks in advance for your collaboration.

closed 2019-09-27 by giunatale