Xilinx/PYNQ #827: sdbuild failed at gcc-mb

riple, 2019-03-07:
I am targeting the ZC702 board, and I ran "make BOARDS=ZC702". My BOARD spec file is as follow

ARCH_ZC702 := arm
BSP_ZC702 := xilinx-zc702-v2018.3-final.bsp 
BITSTREAM_ZC702 := 
STAGE4_PACKAGES_ZC702 := pynq ethernet

I got make failure at the same stage for many times. Here is the last few lines in the make screen print:

cd /home/bibo/work/pynq-zc702/PYNQ/sdbuild/build/gcc-mb && /home/bibo/work/pynq-zc702/PYNQ/sdbuild/packages/gcc-mb//build.sh arm
+ ARCH=arm
+++ dirname /home/bibo/work/pynq-zc702/PYNQ/sdbuild/packages/gcc-mb//build.sh
++ cd /home/bibo/work/pynq-zc702/PYNQ/sdbuild/packages/gcc-mb
++ pwd
+ SCRIPT_DIR=/home/bibo/work/pynq-zc702/PYNQ/sdbuild/packages/gcc-mb
+ case ${ARCH} in
+ sample=arm-linux-gnueabihf,microblazeel-xilinx-elf
+ export -n LD_LIBRARY_PATH
+ ct-ng arm-linux-gnueabihf,microblazeel-xilinx-elf
make[1]: Entering directory '/home/bibo/work/pynq-zc702/PYNQ/sdbuild/build/gcc-mb'
  CONF  config/config.in
#
# configuration written to .config
#

***********************************************************

Initially reported by: Peter Ogden
URL: 

***********************************************************

WARNING! This sample may enable experimental features.
         Please be sure to review the configuration prior
         to building and using your toolchain!
Now, you have been warned!

***********************************************************

Now configured for "arm-linux-gnueabihf,microblazeel-xilinx-elf"
make[1]: Leaving directory '/home/bibo/work/pynq-zc702/PYNQ/sdbuild/build/gcc-mb'
+ ct-ng build
make[1]: Entering directory '/home/bibo/work/pynq-zc702/PYNQ/sdbuild/build/gcc-mb'
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20190307.215921
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_libc_get[scripts/build/libc/newlib.sh@741]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@594]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      '/opt/crosstool-ng/share/doc/crosstool-ng/crosstool-ng-1.22.0/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 1:17.34)
[01:17] / /opt/crosstool-ng/bin/ct-ng:152: recipe for target 'build' failed
make[1]: *** [build] Error 1
make[1]: Leaving directory '/home/bibo/work/pynq-zc702/PYNQ/sdbuild/build/gcc-mb'
/home/bibo/work/pynq-zc702/PYNQ/sdbuild/packages/gcc-mb/Makefile:22: recipe for target '/home/bibo/work/pynq-zc702/PYNQ/sdbuild/build/gcc-mb/arm/microblazeel-xilinx-elf/bin/mb-gcc' failed
make: *** [/home/bibo/work/pynq-zc702/PYNQ/sdbuild/build/gcc-mb/arm/microblazeel-xilinx-elf/bin/mb-gcc] Error 2

riple, 2019-03-07:
The same error happened when I tried "make BOARDS=Pynq-Z1".

yunqu, 2019-03-08:
Check this doc:
https://pynq.readthedocs.io/en/latest/pynq_sd_card.html#building-from-a-board-repository

riple, 2019-03-08:
This is what happened:

bibo@bibo-Inspiron-620:~/work/pynq-zc702/PYNQ/sdbuild$ make BOARDDIR=/home/bibo/work/pynq-zc702/PYNQ/boards/ZC702
which vivado | fgrep 2018.3
/home/bibo/eda/xilinx/Vivado/2018.3/bin/vivado
which sdx | fgrep 2018.3
/home/bibo/eda/xilinx/SDx/2018.3/bin/sdx
which petalinux-config 
/home/bibo/eda/petalinux/tools/common/petalinux/bin/petalinux-config
which arm-linux-gnueabihf-gcc
/home/bibo/eda/xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin/arm-linux-gnueabihf-gcc
which microblaze-xilinx-elf-gcc
/home/bibo/eda/xilinx/SDK/2018.3/gnu/microblaze/lin/bin/microblaze-xilinx-elf-gcc
which ct-ng
/opt/crosstool-ng/bin/ct-ng
which python | fgrep /usr/bin/python
/usr/bin/python
sudo -n mount > /dev/null
bash /home/bibo/work/pynq-zc702/PYNQ/sdbuild/scripts/check_Env.sh
bash /home/bibo/work/pynq-zc702/PYNQ/sdbuild/scripts/check_mounts.sh

yunqu, 2019-03-08:
Try `make clean` since your target has already been built

riple, 2019-03-08:
I have done make clean before make. And it failed everytime at the same stage for Z1\Z2\ZC702.
I have seen success for Linux kernel and Ubuntu rootfs, and it failed afterwards.

yunqu, 2019-03-08:
You may need to use `sudo losetup -a` to check whether a mounted image has been unmounted. 

An dumb way is to reboot the system and run `make clean` then `make BOARDDIR=<yours>`.

riple, 2019-03-11:
I have rebooted my system and Again, after _make clean_, _make _BOARDS=Pynq-Z2__ gives me this:

make[1]: Entering directory '/home/bibo/work/pynq-zc702/PYNQ/sdbuild/build/gcc-mb'
  LN    config
  MKDIR config.gen
  IN    config.gen/arch.in
  IN    config.gen/kernel.in
  IN    config.gen/cc.in
  IN    config.gen/binutils.in
  IN    config.gen/libc.in
  IN    config.gen/debug.in
  CONF  config/config.in
#
# configuration written to .config
#

***********************************************************

Initially reported by: Peter Ogden
URL: 

***********************************************************

WARNING! This sample may enable experimental features.
         Please be sure to review the configuration prior
         to building and using your toolchain!
Now, you have been warned!

***********************************************************

Now configured for "arm-linux-gnueabihf,microblazeel-xilinx-elf"
make[1]: Leaving directory '/home/bibo/work/pynq-zc702/PYNQ/sdbuild/build/gcc-mb'
+ ct-ng build
make[1]: Entering directory '/home/bibo/work/pynq-zc702/PYNQ/sdbuild/build/gcc-mb'
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20190312.061946
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_libc_get[scripts/build/libc/newlib.sh@741]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@594]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      '/opt/crosstool-ng/share/doc/crosstool-ng/crosstool-ng-1.22.0/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 2:54.97)
[02:55] / /opt/crosstool-ng/bin/ct-ng:152: recipe for target 'build' failed
make[1]: *** [build] Error 1
make[1]: Leaving directory '/home/bibo/work/pynq-zc702/PYNQ/sdbuild/build/gcc-mb'
/home/bibo/work/pynq-zc702/PYNQ/sdbuild/packages/gcc-mb/Makefile:22: recipe for target '/home/bibo/work/pynq-zc702/PYNQ/sdbuild/build/gcc-mb/arm/microblazeel-xilinx-elf/bin/mb-gcc' failed
make: *** [/home/bibo/work/pynq-zc702/PYNQ/sdbuild/build/gcc-mb/arm/microblazeel-xilinx-elf/bin/mb-gcc] Error 2

And _make BOARDDIR=Pynq-Z2_ gives me this and does nothing:

which vivado | fgrep 2018.3
/home/bibo/eda/xilinx/Vivado/2018.3/bin/vivado
which sdx | fgrep 2018.3
/home/bibo/eda/xilinx/SDx/2018.3/bin/sdx
which petalinux-config 
/home/bibo/eda/petalinux/tools/common/petalinux/bin/petalinux-config
which arm-linux-gnueabihf-gcc
/home/bibo/eda/xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin/arm-linux-gnueabihf-gcc
which microblaze-xilinx-elf-gcc
/home/bibo/eda/xilinx/SDK/2018.3/gnu/microblaze/lin/bin/microblaze-xilinx-elf-gcc
which ct-ng
/opt/crosstool-ng/bin/ct-ng
which python | fgrep /usr/bin/python
/usr/bin/python
sudo -n mount > /dev/null
bash /home/bibo/work/pynq-zc702/PYNQ/sdbuild/scripts/check_Env.sh
bash /home/bibo/work/pynq-zc702/PYNQ/sdbuild/scripts/check_mounts.sh

yunqu, 2019-03-18:
```
make BOARDDIR=boards
```
where you have `boards/Pynq-Z2`

riple, 2019-03-19:
Hi Yunqu,

My settings are like this:

_bibo@bibo-Inspiron-620:~/work/pynq-zc702/PYNQ/boards/ZC702$ ls
download.bit  xilinx-zc702-v2018.3-final.bsp  ZC702.spec
bibo@bibo-Inspiron-620:~/work/pynq-zc702/PYNQ/boards/ZC702$ cat ZC702.spec 
ARCH_ZC702 := arm
BSP_ZC702 := xilinx-zc702-v2018.3-final.bsp 
BITSTREAM_ZC702 := download.bit 
STAGE4_PACKAGES_ZC702 := pynq ethernet_

yunqu, 2019-03-19:
I think a better way is to put a separate board folder instead of putting it inside your pynq repo.

e.g. 
```
ls /home/bibo/work/pynq-zc702/boards
```
shows only the ZCU702 folder, then 

```
cd /home/bibo/work/pynq-zc702/PYNQ/sdbuild
make BOARDDIR=/home/bibo/work/pynq-zc702/boards
```

In this way you are only building the ZC702 image without building any of the other images.

Going back to the errors you have seen when building gcc-mb, it is probably the ct-ng tools or qemu that gives you the problem. Please make sure you are using the same version of tools as our building flow. Or you may just want to run the setup_host.sh.

wz-heng, 2019-03-21:
I have encountered the same error,I found the source file has an error
in "/opt/crosstool-ng/lib/crosstool-ng-1.22.0/scripts/build/libc/newlib.sh"
this code "{http://mirrors.kernel.org/sourceware/newlib,
                     ftp://sourceware.org/pub/newlib}" 
needs to be changed to "ftp://sourceware.org/pub/newlib" or "http://mirrors.kernel.org/sourceware/newlib"

riple, 2019-03-23:
Thank you, wz-heng! This fix works for me!

It seems that _crosstool-ng_ was installed in _setup_host.sh_, and this patch can be applied in this batch file automatically. Can this be added into the git repository?@yunqu

closed 2019-03-23 by riple