Xilinx/PYNQ #831: fpga_manager fpga0: Error while writing bin data to FPGA with Ultra96

hablu, 2019-03-18:
The error is kind of similar to the #767 but as it has been closed, and my board is different, I am creating this new issue.

[image]
I am facing some error while trying to write a bin file to my Ultra96 board.
I have been using the board image (Avnet Ultra96 v2.4 PYNQ image) from here: 
[image]
Can you please tell me how can I solve it? Thanks.

yunqu, 2019-03-18:
Try to provide an absolute path for your bitstream. This is a different issue than #767 where a customized image needs to be built.

giunatale, 2019-09-27:
Hi @hablu, I see your question was already answered. If you are still not satisfied, I need to kindly ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase.

Closing this now, thanks in advance for your collaboration.

closed 2019-09-27 by giunatale