Xilinx/PYNQ #834: attach to running bitstream

nachiket, 2019-03-21:
Is it possible to attach to a currently running bitstream on the FPGA without triggering a reconfiguration with the `Overlay("bitfile")` command? I'd like to keep a persistent bitstream running, and just DMA stuff on/off as required.

schelleg, 2019-03-21:
Hi @nachiket 

Yes, to query a known overlay with no bitstream download:

```python
ol = Overlay("bitfile", download=False)
```

To query what is currently running on the PL:

```python
from pynq import PL, Overlay
ol = Overlay(PL.bitfile_name, download=False)
```

nachiket, 2019-03-21:
Thanks.. I now seem to get a different error 
```
RuntimeError: DMA channel not started
```

If I try to start the DMA manually
```
>>> overlay.myip.axi_dma.recvchannel.start()
```
its just stuck...

Do I need to do something else to restart DMAs again?

schelleg, 2019-03-21:
There is a good reference here for completing a DMA transfer: [PYNQ-HelloWorld's Resizer Notebook](https://github.com/Xilinx/PYNQ-HelloWorld/blob/master/boards/Pynq-Z1/resizer/notebooks/resizer_PL.ipynb).

Let us know if that series of DMA calls works for you.

nachiket, 2019-03-21:
Yeah that's what I do -- the challenge seems to be calling `run_kernel()` (something similar) a second time causes the DMA error (only a clean reconfiguration restores the DMA).

nachiket, 2019-03-21:
Hmm -- I think its a timing error in my RTL :P

closed 2019-03-21 by nachiket