Xilinx/PYNQ #841: Arduino_IO does not allow mixed input and output pins

rpatel3001, 2019-04-02:
It is currently not possible to concurrently use an Arduino pin as an input while using another Arduino pin as an output. This is because the Arduino_IO class sets the direction register to either 0x0 or 0xffffffff whenever a new object is instantiated.

rpatel3001, 2019-04-02:
#842 solves the original problem, But I now have another related issue. I cannot use two pins as output simultaneously. Creating multiple Arduino_IO objects set to output results in only the last one created actually being functional. The others accept write() commands but do not physically change state. 

Creating multiple inputs works properly. I will look into a fix and add it to #842.

schelleg, 2019-04-04:
thank you for the contribution!

closed 2019-04-04 by schelleg

yunqu, 2019-04-04:
Although this is close, I want to add that the Arduino_IO class should enable read-modify-write for IO switch configuration, otherwise the old object will get disturbed by the new object on a different pin index. In short, this issue is still not fully resolved.

rpatel3001, 2019-04-05:
@yunqu I'll let you and @schelleg hash it out in new issue. I've found a solution to my problem and don't have the time to keep up with this.

schelleg, 2019-04-05:
I committed onto @rpatel3001 's PR to make the Arduino IO Microblaze a singleton.   I believe this resolved @yunqu 's concern.  See my testing gist that verified the RMW behavior:
https://gist.github.com/schelleg/9d78a4b72032cde9133c6c6297514d9b

rpatel3001, 2019-04-05:
I think @yunqu wants the iop_switch_config member of Arduino_DevMode class to be read-modify-write. As is it gets reset to ARDUINO_SWCFG_DIOALL when the singleton is initialized.