Xilinx/PYNQ #845: DMA recvchannel stop working after use DMA.recvchannel.transfer(output_buffer)

eleICoto, 2019-04-07:
It seems like my dma recvchannel stop running after i run DMA.recvchannel.transfer(output_buffer) 
I'm using .hwh and i think my .bit is ok because dma works fine in sdk.

eleICoto, 2019-04-08:
By the way i'm using zcu104

eleICoto, 2019-04-08:
The .bit file i'm using is like this
[image]
[image]
[image]

eleICoto, 2019-04-08:
The error in python is like this
[image]

eleICoto, 2019-04-08:
The code i'm using is like this 
[image]

yunqu, 2019-04-08:
I think the execution order matters. See our example at: https://github.com/Xilinx/PYNQ-HelloWorld/blob/master/boards/ZCU104/resizer/notebooks/resizer_PL.ipynb

where you want to do something like
```python
dma.sendchannel.transfer(in_buffer)
dma.recvchannel.transfer(out_buffer)    
 resizer.write(0x00,0x81) # start
 dma.sendchannel.wait()
 dma.recvchannel.wait()
```

eleICoto, 2019-04-09:
[image]
[image]
I think i have tried every combination of execution order, but still the recvchannel stop running after i strat transfer action.

eleICoto, 2019-04-10:
[image]
The zynq setting is like this, i didn't enanble GPIO at the first version.

theDweeb, 2019-05-03:
Hey did you ever figure this issue out? If I use wait_async() or no wait at all it completes the process but the output buffer doesnt have all correct values. If I use wait() it gives me the same error you have

eleICoto, 2019-05-05:
you mean you can't use wait() function but you can still get correct data in buffer?
after I failed with python interface I switch to petalinux

PeterOgden, 2019-05-07:
Would you be able to print the output of `pl2ps.read(0x30)` and `pl2ps.read(0x34)` after the wait fails? It looks like for some reason the DMA engine is erroring out and shutting down but I don't understand why.

Only things that springs to mind is whether TLAST is being asserted on the correct beat - if more data is sent than the DMA is expecting that could cause the error you are seeing, or if the physical_address is getting assigned incorrectly.

theDweeb, 2019-05-07:
I solved my DMA issue by looking into the parameters of the IP and adjusting the width of the buffer length (default 14) to 17. Progress! My new issue is data loss that appears to be a pattern. My IP uses 8 bit data streaming(I’ve tried char and ap_int<8>) but it only streams all of the correct values If the dtype is int64 when I initialize the in_buffer on jupyter notebook. If I set it to int32 it skips every other(places a 0), if it’s int16 it skips two shows two, int8 skips 4 shows 4

Example data coming in: 
[1,2,3,4,5,6,7,8,9]

Int8 output
[1,2,3,4,0,0,0,0,9]

Int16
[1,2,0,0,5,6,0,0,9]

Int32
[1,0,3,0,5,0,7,0,9]

Int64 shows them all correctly

yunqu, 2019-05-07:
I am surprised to see the buffer length of 14 did not raise any warning / error message for you. Did you use tcl or hwh file for your overlay?

theDweeb, 2019-05-07:
just tcl at the moment. My Z1 board is still using pynq 2.2. I was just recently made aware of the hwh when I gained access to an ultra96 running the pynq 2.4 image. I plan on reflashing my Z1 to 2.4 when I get home.

I am trying to get a basic 4 layer MLP using 5bit quantized weights for mnist 28x28 handwritten digits.

The first hidden layer has 100 neurons so the first multiplication is 784x100 5 bit numbers (switched to 8bit since I do not know how to set the dtype for the cma.numpy array to 5)

After wondering why my transpose worked on the host but not in the hardware I noticed the patterned data loss I mention in my previous post

eleICoto, 2019-05-07:
So, why 17, is there any reason behind it?
And, I'm wondering will it be the problem of my address width setting?
I look into the resize example of ZCU104, it shows that the address width is set to 32

yunqu, 2019-05-07:
@eleICoto This is the buffer length width, so 17 means your DMA buffer can be as big as 2^17 = 128K bytes. For resizer example that is set to the maximum value, 4GB.

yunqu, 2019-05-07:
@theDweeb Better to post in another thread on pynq support forum. We are discouraged to discuss issues here.:)

theDweeb, 2019-05-07:
Ok sorry. The pynq forums do not load for me (or any Google groups for that matter)

yes 784*100 at 8 bit values = 78400 bytes. log2(78400) is 16.25 so I rounded up to 17.

I should clarify that I am not doing your resizer but had similar issues (I just googled my problem and this thread popped up)

When using 2.4 without hwh I get a big warning. with hwh I receive no warning.

thanks again!

eleICoto, 2019-05-07:
@yunqu Alright got it, depends on buffer size.
So do you have any idea on address width?

yunqu, 2019-05-07:
@eleICoto See my previous post

eleICoto, 2019-05-07:
@yunqu Emmmm, I'm sorry, previous post?

eleICoto, 2019-05-07:
@theDweeb So the wait() function is working correctly?

yunqu, 2019-05-10:
We have new support forum so you do not have to use google forum any more. If you have further questions, please post here: https://discuss.pynq.io/c/support

closed 2019-05-10 by yunqu