Xilinx/PYNQ #846: KeyboardInterrupt dma.sendchannel.wait()

HungSoma, 2019-04-08:
Hi,
Im trying to use DMA to tranfer an image to crop function but when dma.sendchannel.wait(), kernel always excute and I interupt it that error appear. 
Pls help me !
Thanks all!
[image]
[image]
[image]
[image]

yunqu, 2019-04-08:
What is the size of `x`? Also, it looks like your `in_buffer` is 640x480x3 while your instantiate DMA buffer as 640x427x3. Not sure if this is going to be a problem.

HungSoma, 2019-04-09:
x.size = ( 427, 640 ). I changed to dma.sendchannel.wait_async and dma.recvchannel.wait_async and it work  but result not as expect:
[image]
what  is the different between dma.sendchannel.wait() and dma.sendchannel.wait_async() ?

giunatale, 2019-09-27:
Hi @HungSoma, I see your question was already answered. If you are still not satisfied, I need to kindly ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase.

Closing this now, thanks in advance for your collaboration.

closed 2019-09-27 by giunatale