Xilinx/PYNQ #851: Cannot log into Jupyter on Firefox 66

c-x-berger, 2019-04-18:
I just burnt image version 2.4 onto my SD card and booted for the first time today. I connected an Ethernet cable from the Z1 to my laptop, set up a static IP on said laptop, and pointed Firefox at http://192.168.2.99:9090. I was greeted with a simple login screen:
[image]
So I put in the default password of `xilinx`. That simply redirected to an identical page. I could put in the password a seemingly infinite number of time with no change.

After browsing the issues here, I noticed some people had issues logging in on specific browsers, so I tried opening the page in Chromium. Putting in the exact same password (`xilinx`) logged me in just fine:
[image]

cathalmccabe, 2019-04-18:
This is a known issue that we hope to have a better workaround for in the next release. The issue is the board doesn’t have a real-time clock, so the first time you connect to the board with a direct connection (direct Ethernet connection to laptop) the board time will be a long way out of date. (It will be set to the image build time.) This causes problems with authentication cookies on some browsers. The problem will occur if the board time is more than ~30 days out of date compared to your PC. 
One workaround is to log in to your board and update the time, then use your preferred browser. (No judging!)
The time is saved periodically, so when you power off your board it will restore the most recent time, so if you have your board on a lot it will take a while before it drifts out of sync again. 
Another workaround is to connect the board to your home router, and if it has internet access, it should auto update time from the internet. 

By the way, the workaround is to set the cookie authentication time to be a lot longer than 30 days which we plan to do in the next release.

c-x-berger, 2019-04-20:
OK. I set the time on the board to be correct and everything seems to be mostly alright. Thanks!

closed 2019-05-07 by yunqu