Xilinx/PYNQ #853: Pynq Z-1, notebook crash whenn calling dma

mukechiltmal, 2019-05-01:
Hey there,
i have been browsing for quite some and still havent found a solution to my issue
The system i use is the following:
Pynq Z1 with v2.4 image
A custom overlay with base overlay as template
as target system the pynq z1 board files

now to my issue:
In my overlay i am trying to use a dma to stream data from ps to an accelerator in pl and after processing back to ps
whenever i want to create an object of the said dma in the jupiter notebook, the system gets stuck/ crashes and requires a hard reset

here my DMA setup in vivado

[image]

a screenshot of the dma part of the block design

[image]

the used notebook

[image]

for anymore information let me know,

thanks
mukechiltmal

cathalmccabe, 2019-05-01:
Can you post to the support forum please? 
http://www.pynq.io/support.html

mukechiltmal, 2019-05-01:
sure, was waiting to get access xD
Closed

closed 2019-05-01 by mukechiltmal