Xilinx/PYNQ #870: I have a problem when running setup_host.sh to build PYNQ image

toannv123, 2019-06-20:
When running setup_host.sh to build PYNQ image, I faced this problem. 

util/memfd.c:40:12: error: static declaration of ‘memfd_create’ follows non-static declaration
 static int memfd_create(const char *name, unsigned int flags)
            ^~~~~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/bits/mman-linux.h:115:0,
                 from /usr/include/x86_64-linux-gnu/bits/mman.h:45,
                 from /usr/include/x86_64-linux-gnu/sys/mman.h:41,
                 from /home/toan/pynq/sdbuild/scripts/tools/qemu-2.8.0/include/sysemu/os-posix.h:29,
                 from /home/toan/pynq/sdbuild/scripts/tools/qemu-2.8.0/include/qemu/osdep.h:104,
                 from util/memfd.c:28:
/usr/include/x86_64-linux-gnu/bits/mman-shared.h:46:5: note: previous declaration of ‘memfd_create’ was here
 int memfd_create (const char *__name, unsigned int __flags) __THROW;
     ^~~~~~~~~~~~
/home/toan/pynq/sdbuild/scripts/tools/qemu-2.8.0/rules.mak:60: recipe for target 'util/memfd.o' failed
make: *** [util/memfd.o] Error 1

toannv123, 2019-06-20:
OS: Ubuntu 18.04.2 LTS

giunatale, 2019-06-20:
Hi,

After a quick google search of your issue, turns out it is a common problem found when using qemu on ubuntu 18.04.
As this is not strictly related to PYNQ, the quick solution I would suggest is to run the build process on Ubuntu 16.04 instead. You can build a VM using Vagrant and the Vagrantfile provided (https://github.com/Xilinx/PYNQ/blob/master/Vagrantfile).

Best,
Giuseppe Natale

EXTERNAL EMAIL

toannv123, 2019-06-21:
Thank you for your quick reply. I fixed it by using patch file.

https://patchwork.ozlabs.org/patch/842091/

closed 2019-06-24 by giunatale