Xilinx/PYNQ #874: Error while writing image data to FPGA

ted51, 2019-07-10:
When write a bit stream .bin file to FPGA via FPGA manager, I met this failure:
[ 1758.505854] fpga_manager fpga0: writing 1x16x16_8bx8b_15_15_18_17_100MHz_8ns_v0_0_0.bin to Xilinx ZynqMP FPGA Manager
[ 1758.588004] fpga_manager fpga0: Error while writing image data to FPGA
[ 1758.597566] fpga_manager fpga0: writing 1x16x16_8bx8b_15_15_18_17_100MHz_8ns_v0_0_0.bin to Xilinx ZynqMP FPGA Manager
[ 1758.680143] fpga_manager fpga0: Error while writing image data to FPGA

I'm using board ZCU104 and pre-built PYNQ image "ZCU104 v2.3 SDCard image" released in https://github.com/Xilinx/PYNQ/releases

I saw a similiar issue #767(https://github.com/Xilinx/PYNQ/issues/767) and it's fixed by a patch. I checked the git log of PYNQ reponsitory and found this patch was already merged in ZCU104 v2.3 image. 
I also checked the ZCU104 v2.4 image and had the same issue.

Could somebody help to give some suggestions?
Any comments is appreciated.

rainfld, 2019-08-14:
Hi, @ted51, any updates on this issue? I got the exact same issue with you. I'm using pre-built PYNQ image "ZCU104 v2.4 SDCard image". Any suggestion is appreciated.

giunatale, 2019-09-27:
Hi @ted51 and @rainfld, I don't know if you got around this issue, but in any case I need to kindly ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase.

Closing this now, thanks in advance for your collaboration.

closed 2019-09-27 by giunatale