Xilinx/PYNQ #876: Setting operating frequency of PS to something else than 650 MHz

bluesceada, 2019-07-16:
The frequency of the PS (Dual Core ARM Cortex-A9 in Pynq Z1) seems to be not changeable. There seems to be no working cpufreq governor, and /proc/cpuinfo only seems to report the MHz as "BogoMIPS" (?).

Furthermore, when trying the python3 pynq package as root it leads to the following:
>>> from pynq import ps
>>> ps.Clocks.cpu_mhz = 600.00
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/xilinx/pynq/ps.py", line 78, in cpu_mhz
    raise RuntimeError("Not allowed to change CPU clock.")
RuntimeError: Not allowed to change CPU clock.
>>>

Please clarify how it is possible to use the PS with another frequency (even if just statically). Is there maybe a way by changing the Linux kernel commandline (CMD)? Thanks!

jkrautter, 2019-09-09:
You can add `CONFIG_CPUFREQ_DT=y` to `sdbuild/boot/meta-pynq/recipes-kernel/linux/linux-xlnx/pynq.cfg` and recompile BOOT.BIN. This enables cpufreq to change the ARM core frequencies between 650 and 325 MHz. 
If you need a specific frequency other than 650 and 325 MHz, I managed to do that by modifying the base designs `boards/Pynq-Z1/base/base.tcl` and `boards/Pynq-Z1/petalinux_bsp/hardware_project/pynqz1.tcl`. You'll have to edit the frequencies as well as the PLL configurations. Additionally, I had to overlay the originally defined operating points of the CPU in the device tree for linux to boot properly. This can be done by adding the following to `boards/Pynq-Z1/petalinux_bsp/meta-user/recipes-bsp/device-tree/files/system-user.dts`:
```
cpus {
       cpu@0 {
          operating-points = <
             123456  1000000
          >;
       };
    };
```
This has to be placed before the `amba` block and 123456 should be the desired frequency in kHz.

giunatale, 2019-09-27:
Hi @bluesceada, I see your question was already answered. If you are still not satisfied, I need to kindly ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase.

Closing this now, thanks in advance for your collaboration.

closed 2019-09-27 by giunatale